Bodmer/TFT_eSPI #1880: Drawing White Line Instead of Green Line

fervenceslau, 2022-06-19:
```
To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
```
When I try to print a green horizontal line segment on rows 0 or 1 using either draw line functions I get a white line instead. I tried with the colors red and blue and they seem to work. Also, when drawing a rect to cover the whole screen it does work properly. Is it possible that I received a faulty display? 

I tried changing the defitinitions ins user_setup.h, but the problem still persists. The configuration that worked best using other demo codes was `#define ST7735_REDTAB`.

`2. IDE (e.g. Arduino or PlatformIO)`
Platformio

`3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu`
lib_deps = bodmer/TFT_eSPI@^2.4.70

`4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu`
board = esp32doit-devkit-v1

`6. Procesor, e.g RP2040, ESP32 S3 etc`
ESP32

`7. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.`
Should be a TFT ST7735
https://www.aliexpress.com/item/32817839166.html?spm=a2g0o.order_list.0.0.45201802aDdI9r

`8. Interface type (SPI or parallel)`
SPI

```
Plus further information as appropriate to the problem:
1. TFT to processor connections used
```
#define TFT_MOSI 2 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 15
#define TFT_CS   16  // Chip select control pin
#define TFT_DC   4  // Data Command control pin
#define TFT_RST  5  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL   17  // LED back-light

```
3. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
4. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
```
[TFT_SPI.zip](https://github.com/Bodmer/TFT_eSPI/files/8934798/TFT_SPI.zip)

`5. Screen shot pictures showing the problem  (just drag and drop in message window)`
[image]

Bodmer, 2022-06-19:
I suspect this is down to offsets in the display as the controller is designed for 132 x 162 so there are spare pixel lines and the position of these varies betwen suppliers. Also the display does not work properly with high SPI clok rates.

To test for offsets run this sketch, change it for rotations 1-3 as well. A full green rectangle should be visible at all screen edges in all rotations with no missing sides.
```
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

void setup() {
  tft.begin();
  tft.setRotation(0);
  tft.fillScreen(TFT_BLACK);
  tft.drawRect(0, 0, tft.width(), tft.height(), TFT_GREEN);
}

void loop() {

}
```

fervenceslau, 2022-06-19:
The code you provided works without a problem. The issue only happens when I use drawLine (or equivalent) for the first two rows using green color, but it disappear when you continue to draw other things I think. 

This is the result I get running the code shown below, which prints a green horizontal line for each line.
https://user-images.githubusercontent.com/49814646/174491849-6194de90-0fa2-4410-92f2-1116b37bfd9d.mp4

```
#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup() {
  tft.init();
  tft.setRotation(3);
  tft.fillScreen(TFT_BLACK);
  for (int i=0; i<tft.height(); i++) {
    tft.drawLine(0, i, tft.width() - 1, i, TFT_GREEN);
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}

void loop() { 
}
```

aord, 2022-06-19:
Did you try all the other TAB colours?
My screen had a green tab but worked with used #define ST7735_BLACKTAB

there is also:
#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

fervenceslau, 2022-06-19:
Yeah I tried all TAB colors, the one that worked best was RED, but it looks like BLACK works as well - I would have to try more complex tests to see which is better. Also, other colors also appear to have this issue like yellow, dark green, orange, brown and many more. I think everything is related to the green colors, but I have no idea how to pinpoint what is causing the problem...

Bodmer, 2022-06-20:
This is a rather odd fault. I do not think it is library related asdrawing line 3 green would not make lines 0 and one change colour.

It looks like a display fault, do you have another display of the same type to try?

fervenceslau, 2022-06-20:
Unfortunately I do not have another one. Yeah I also believe its most likely a device problem given the weird behavior.

closed 2022-06-20 by fervenceslau