Bodmer/TFT_eSPI #1882: Sender wise Data Fetching issue in the Receiver to TFT Display

AjitN111, 2022-06-20:
We are facing issue Sender wise Data Fetching to receiver TFT module, Its take arbitrary value .
Please help to resolve the issue 

Code as below

/*
  Rui Santos
  Complete project details at https://RandomNerdTutorials.com/esp32-esp-now-wi-fi-web-server/
  
  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files.
  
  The above copyright notice and this permission notice shall be included in all
  copies or substantial portions of the Software.
*/

#include <esp_now.h>
#include <WiFi.h>
#include "ESPAsyncWebServer.h"
#include <Arduino_JSON.h>
#include <HTTPClient.h>
#include <SPI.h>
#include "MCUFRIEND_kbv.h"
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <TFT_eSPI.h> 
TFT_eSPI tft = TFT_eSPI(); 

#define ESP32_PARALLEL
#define ILI9486_DRIVER
String temperature;
String humidity;

// Replace with your network credentials (STATION)
const char* ssid = "realme";
const char* password = "12345678";

//ACCESS POINT credentials
const char* ssidAP = "ESP_E63B4D";
const char* passwordAP = "";

// Structure example to receive data
// Must match the sender structure
typedef struct struct_message {
  int id;
float temperature; 
float humidity;
unsigned int readingId;
} struct_message;

struct_message incomingReadings;
struct_message incomingReadings1;
struct_message incomingReadings3;

#define TFT_CD D15 
#define TFT_RTS D32
#define TFTWIDTH   480
#define TFTHIGHT   320

#define TFT_BLACK       0x0000      /*   0,   0,   0 */
#define TFT_NAVY        0x000F      /*   0,   0, 128 */
#define TFT_DARKGREEN   0x03E0      /*   0, 128,   0 */
#define TFT_DARKCYAN    0x03EF      /*   0, 128, 128 */
#define TFT_MAROON      0x7800      /* 128,   0,   0 */
#define TFT_PURPLE      0x780F      /* 128,   0, 128 */
#define TFT_OLIVE       0x7BE0      /* 128, 128,   0 */
#define TFT_LIGHTGREY   0xD69A      /* 211, 211, 211 */
#define TFT_DARKGREY    0x7BEF      /* 128, 128, 128 */
#define TFT_BLUE        0x001F      /*   0,   0, 255 */
#define TFT_GREEN       0x07E0      /*   0, 255,   0 */
[... 413 lines cut ...]
  WiFiClient client;
  HTTPClient http;
    
  // Your Domain name with URL path or IP address with path
  http.begin(client, serverName);
  
  // Send HTTP POST request
  int httpResponseCode = http.GET();
  
  String payload = "--"; 
  
  if (httpResponseCode>0) {
    Serial.print("HTTP Response code: ");
    Serial.println(httpResponseCode);
    payload = http.getString();
  }
  else {
    Serial.print("Error code: ");
    Serial.println(httpResponseCode);
  }
  // Free resources
  http.end();

  return payload;
}*/

Bodmer, 2022-06-21:
I am unable to help due to other priorities.

closed 2022-06-21 by Bodmer

AjitN111, 2022-06-21:
Okey no issue . Thank you