Bodmer/TFT_eSPI #1884: TFT_eSPI/examples/Sprite/Animated_dial - slight corruption with jpg background loaded from SD

Frank-Bemelman, 2022-06-22:
First, thank you for your inspiring work, it is highly appreciated. I am toying around with an old jukebox and used your dial example which works brilliant. But since I use it for a volume indicator when setting the volume with a rotary encoder knob, I wanted to use a different background image which goes from 0-100 - makes more sense that 0-240 in my situation.

Found a dial on the internet, scaled it to 240x240 which is the viewable area of my 320x240 ILI9341 display. 
Stored it on the SD card, and draw it on the display. So far so good.

```
TJpgDec.drawSdJpg(0, 0, "/dial2.jpg");

```

But when I move the needle around, it slightly corrupts the divisions around the knob. 
I think it has something to do with the 24 bit color depth of the jpg, which gets converted to 16 bit color depth when small portions of the display are fetched and  pushed back between the needle movements.

[image]

Not sure how to fix this. How did you create the array in dial.h? That might be better than loading it off the SD card, as it seems to be a bit faster as well.

Thank you for suggestions,
Cheers, Frank

Bodmer, 2022-06-22:
It looks like read/write data corruption. Yo may have to reduce the read and or write clock rates in the setup file.

Info on file conversion is in the header comments here:
https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Sprite/Animated_dial/dial.h

Frank-Bemelman, 2022-06-22:
Wow! Exactly that! I reduced the read frequency by half:
#define SPI_READ_FREQUENCY  10000000

The distortions were more obvious with a more colored background, but now it is perfectly clean:

[image]

Thank you very much - my wires to the display are quite long, some 40cm long, and the entire contraption is quit messy. I will certainly give it a try to clean that up a little, and see if I can increase the clock speed back to default. I had absolutely no problems with the displaying side of things, so I never worried about my dangling wires being part of the problem,

Thanks again ;-)
Cheers.
Frank

Bodmer, 2022-06-22:
Great, glad it is working.