Bodmer/TFT_eSPI #1894: 4.0 inch SPI module 3.95 inch ILI9488 14 pin TFT LCD display, grid issue

fungreenfox, 2022-06-30:
I got a 4.0" spi module, 3.95 inch ili9488 with some grid issues. 
It is mounted to an esp32, with Bodmers suggestion. It works, but the bottom edge vertical way, rotation 3, is missing the grid from edge to at ca 4mm way in,all alongside, where it does not registrate touch.
When running the calibration, (the one that will paint a dot where pen is set), it works all over the display.
The display is setup to ili9488 with width 480 and hight 320.
Can this be a setup issue?
https://www.aliexpress.com/item/1005003033828159.html?spm=a2g0o.order_list.0.0.10625c5faGZVEU

Bodmer, 2022-07-17:
Make sure the calibration is done in the screen rotation that you use in you own sketch.

Note that some touch screen screens has a small zone that is outside the TFT pixel area, this is visible as the black strip in the picture below on the left side:
[image]

If this does not answer the problem then please provide a sletch that demonstrates the problem.

closed 2022-07-17 by Bodmer

Bodmer, 2022-07-17:
Another possibility is that the touch screen is insensitive near the edge for touch pressure. Run the Test_Touch_Controller sketch and look at the z value when touching the screen area you are having trouble with. The default pressure threshold for a valid touch value is 600. You can alter this threshold using an extra parameter in the call, for example:

tft.getTouch(&t_x, &t_y, 250)

Sets the pressure detection threshold to 250.

fungreenfox, 2022-11-11:
> Another possibility is that the touch screen is insensitive near the edge for touch pressure. Run the Test_Touch_Controller sketch and look at the z value when touching the screen area you are [...]

Worth trying out, thank You.

fungreenfox, 2022-11-12:
But if the touch at the edge makes it mark dots 9mm into the area, then its not the touch depth, but a fabric fault?