Bodmer/TFT_eSPI #1902: ESP32 UNO with TFT SHIELD ILI9486 PARALLEL

kwittek, 2022-07-08:
Hi,
i try to use PARALLEL connection with ILI9486.  Is it possible? 
I only get white screen during TFT_Print_Test, result of Read_User_Setup looks correctly:

01:29:56.727 -> [code]
01:29:56.727 -> TFT_eSPI ver = 2.4.61
01:29:56.727 -> Processor    = ESP32
01:29:56.727 -> Frequency    = 240MHz
01:29:56.727 -> Transactions = Yes
01:29:56.727 -> Interface    = Parallel
01:29:56.727 -> Display driver = 9486
01:29:56.727 -> Display width  = 320
01:29:56.727 -> Display height = 480

I solder additional wires. 
This is how my user_setup.h file looks like: 

// User defined information reported by "Read_User_Setup" test & diagnostics example
#define USER_SETUP_INFO "User_Setup"
//#define USER_SETUP_ID 14

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
#define TFT_PARALLEL_8_BIT

#define ILI9486_DRIVER

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
#define TFT_PARALLEL_8_BIT

// ESP32 pins used for the parallel interface TFT
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

 #define SPI_FREQUENCY   1000000
// #define SPI_FREQUENCY   5000000
// #define SPI_FREQUENCY  10000000

//#define SPI_READ_FREQUENCY  20000000

Any ideas what should i change? 

Thanks in advance

Bodmer, 2022-07-08:
Have you added the wire links described here:
https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-support

kwittek, 2022-07-08:
> Have you added the wire links described here: https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-support

Thanks for replay, my display was broken. I change it to another and it works. Thanks for replay anyway. Please delete this issue, it will not help anyone

closed 2022-07-08 by Bodmer