Bodmer/TFT_eSPI #1904: C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

yeodeanyeo, 2022-07-09:
To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
2. IDE (e.g. Arduino or PlatformIO)
3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
5. Procesor, e.g RP2040, ESP32 S3 etc
6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
7. Interface type (SPI or parallel)

Plus further information as appropriate to the problem:
1. TFT to processor connections used
2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
4. Screen shot pictures showing the problem  (just drag and drop in message window)

The idea is to provide sufficient information so I can setup the exact same (or sufficiently similar) scenario to investigate and resolve the issue.

Put three ticks ( ` character on key next to "1" key) at the start and end of short pasted code segments to avoid format/markup anomolies and to use a monspaced font. Example:

```
  Serial.begin(115200);
  tft.init();
```

yeodeanyeo, 2022-07-09:
C:\Users\Yeo Dean\Documents\Arduino\ESP32_ILI9341_MY_Menu\ESP32_ILI9341_MY_Menu.ino: In function 'void setup()':
C:\Users\Yeo Dean\Documents\Arduino\ESP32_ILI9341_MY_Menu\ESP32_ILI9341_MY_Menu.ino:361:54: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   getstarted.initButton(17.5,240,205,30,"Get Started");     //First Button to bring up 2nd Screen
                                                      ^
C:\Users\Yeo Dean\Documents\Arduino\ESP32_ILI9341_MY_Menu\ESP32_ILI9341_MY_Menu.ino: At global scope:
ESP32_ILI9341_MY_Menu:367:3: error: 'buttonjump' does not name a type
   buttonjump.initButton(17.5,120,100,30,"JUMP");
   ^~~~~~~~~~
C:\Users\Yeo Dean\Documents\Arduino\ESP32_ILI9341_MY_Menu\ESP32_ILI9341_MY_Menu.ino: In function 'void loop()':
C:\Users\Yeo Dean\Documents\Arduino\ESP32_ILI9341_MY_Menu\ESP32_ILI9341_MY_Menu.ino:426:45: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   buttonwifi.initButton(80,dist,80,30,"WIFI");
                                             ^
C:\Users\Yeo Dean\Documents\Arduino\ESP32_ILI9341_MY_Menu\ESP32_ILI9341_MY_Menu.ino:460:51: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   buttontemp.initButton(17.5,240,205,30,"GET TEMP"); //"TEMP" fits just nice in the colored box
                                                   ^
Using library SPI at version 2.0.0 in folder: C:\Users\Yeo Dean\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\SPI 
Using library Adafruit_GFX_Library at version 1.11.1 in folder: C:\Users\Yeo Dean\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Adafruit_ILI9341 at version 1.5.11 in folder: C:\Users\Yeo Dean\Documents\Arduino\libraries\Adafruit_ILI9341 
Using library ThingPulse_XPT2046_Touch at version 1.4 in folder: C:\Users\Yeo Dean\Documents\Arduino\libraries\ThingPulse_XPT2046_Touch 
Using library TouchEvent at version 1.3 in folder: C:\Users\Yeo Dean\Documents\Arduino\libraries\TouchEvent 
Using library Adafruit_MLX90614_Library at version 2.1.3 in folder: C:\Users\Yeo Dean\Documents\Arduino\libraries\Adafruit_MLX90614_Library 
Using library Adafruit_BusIO at version 1.11.3 in folder: C:\Users\Yeo Dean\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 2.0.0 in folder: C:\Users\Yeo Dean\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire 
Using library WiFiManager at version 2.0.11-beta in folder: C:\Users\Yeo Dean\Documents\Arduino\libraries\WiFiManager 
Using library WiFi at version 2.0.0 in folder: C:\Users\Yeo Dean\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi 
Using library Update at version 2.0.0 in folder: C:\Users\Yeo Dean\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Update 
Using library WebServer at version 2.0.0 in folder: C:\Users\Yeo Dean\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WebServer 
Using library DNSServer at version 2.0.0 in folder: C:\Users\Yeo Dean\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\DNSServer 
Using library FS at version 2.0.0 in folder: C:\Users\Yeo Dean\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\FS 
exit status 1
'buttonjump' does not name a type

Bodmer, 2022-07-10:
This repository is for TFT_eSPI only. Your sketch uses the Adafruit library, which I do not provide support.

closed 2022-07-10 by Bodmer