Bodmer/TFT_eSPI #1905: Share ILI9341 display with MAX31855 on the same SPI bus

jankovalski, 2022-07-10:
I don't think this is a problem with tft_espi itself but I have no idea where to ask. I have a project based on ESP32 Dev Module, ADS1115 i2c ADCs and ILI9341 touch screen. It's been working great for a long time now. But recently I added MAX31855 thermocouple IC, one the same SPI bus as the LCD, to read EGT sensor. It is working great when it's the only thing attached to the ESP32, it can reach 1600 loops per second and the readings will remain stable. But when I have it all in one loop, MAX31855 readings will become unstable when I'm above 8-9 loops per second. I attached a video. The 23°C is the correct EGT value and the number above it, is the calculated loops per second value. When it's at 10, the EGT readings are unstable. From what I can see, 1 out of 3 readings is correct. But when I decrease ADS1115 sampling rate, therefore I slow down the entire loop and the loop happens only 7 times per second, the EGT reading are stable again. The whole code is happening in a single loop, within single core, so the tft code happens only when MAX31855 code is done and the other way around. Nothing happens in parallel here. I've tried everything, messed with SPI speeds in User_Setup.h, moved both devices from VSPI (which is what I'm using) to HSPI, tried different libraries for MAX31855... Nothing helps. Even if I add a 1s delay after reading MAX31855, it still will be unstable, just slower... Here's my User_Setup:

```
#define TFT_DC    2
#define TFT_RST   4
#define TFT_CS   15
#define TFT_SCLK 18
#define TFT_MOSI 23
#define TOUCH_CS  5
#define TOUCH_DO 19
```

MAX31855 MISO is attached to pin 19, SCK to pin 18 and CS to pin 27.

ESP32 Dev Module
tft_espi latest version

Do you have any ideas? Is there maybe a void/bool that I can use before reading MAX31855 to check if SPI isn't currently in use by tft_espi?

https://user-images.githubusercontent.com/46295168/178158274-304ea4a3-70c1-47d1-9716-237cd1785b6e.mp4

CETan655, 2022-07-11:
jankovalski, the Max31855 has a few specifications that may constrain you. 
1) The temperature conversion may take up to 100ms, so it is definitely not able measure more than 10 Hz. 
2) The max serial clock frequency is only 5Mhz. 
3) The  serial data output from Max31855 is 32bit as oppose to the usual 8bit format use in hardware SPI. Please refer to the MAX31855 data sheet for further information
[image]
[image]
[image]
.

Bodmer, 2022-07-17:
I've recently started using the MAX31855 with the Adafruit_MAX31855 library and have not seen any issues.

I created this simple sketch and it runs fine:
```
#include <SPI.h>
#include "Adafruit_MAX31855.h"
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

#define MAXCS   21

Adafruit_MAX31855 thermocouple(MAXCS);

void setup() {
  Serial.begin(115200);

  while (!Serial) delay(1); // wait for Serial

  tft.init();
  tft.setRotation(3);
  tft.fillScreen(TFT_BLACK);
  tft.setTextSize(4);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);

  Serial.println("MAX31855 test");
  // wait for MAX chip to stabilize
  delay(500);

  Serial.print("Initializing sensor...");
  if (!thermocouple.begin()) {
    Serial.println("ERROR.");
    while (1) delay(10);
  }
  Serial.println("DONE.");
}

void loop() {
   float c = thermocouple.readCelsius();
   if (isnan(c)) {
     Serial.println("Something wrong with thermocouple!");
   } else {
     Serial.println(c);
     tft.setCursor(0, 0);
     tft.println(c);
   }
   delay(50);
}
```

To help further I will need an example sketch that demonstrates the problem. Otherwaise I can only guess at the cause, for example, if you are using DMA to the screen then you must wait for that to complete before accessing the MAX31855.

closed 2022-07-19 by Bodmer