Bodmer/TFT_eSPI #1908: ESP32 v2.0.1 and later error on init()

JimDrewGH, 2022-07-12:
I upgraded from v1.0.7 of the ESP32 framework to v2.0.4.  When I power up after flashing, I get this error message in the serial monitor:

E (139) gpio: gpio_set_level(226): GPIO output gpio_num error

I traced this down literally to just having:

tft.init(); 

Just having this enabled generates this error.  I do have a custom config file (shown at the end of this text).  Everything seems to work fine, but this message popping up in the serial monitor every time I boot is a bit alarming.

I am using Arduino IDE 1.8.19.

I am using the latest TFT_eSPI (v2.4.61 as installed by the library manager).

I tested this with v2.0.1 and later ESP32 framework.  I did not test v2.0.0, but v1.0.6 and earlier does not generate the message.

This is being tested on a ESP32 WROOM module.

This is a ST7735, 1.8" TFT (160x128) display.  Using the SPI bus.

This is the entire sketch that I am testing with:

`#include <SPI.h>
#include <TFT_eSPI.h>                         // Hardware-specific library

// SETUP

TFT_eSPI tft = TFT_eSPI();                    // Invoke custom library

void setup() {
  Serial.begin(115200);
  delay(10);                                  // allow serial to initialize
  Serial.println("Starting up!");
  tft.init();                                    // <<----- here is the line to init the display that generates the error
  Serial.println("Done!");
}

void loop() {
}
`

Here is the custom config file:

`// See SetupX_Template.h for all options available

#define ST7735_DRIVER
//#define ST7789_DRIVER
//#define ILI9341_DRIVER 

#define TFT_WIDTH  128
#define TFT_HEIGHT 160

// #define ST7735_INITB
// #define ST7735_GREENTAB
// #define ST7735_GREENTAB2
// #define ST7735_GREENTAB3
// #define ST7735_REDTAB
 #define ST7735_BLACKTAB

// If colours are inverted (white shows as black) then uncomment one of the next
// 2 lines try both options, one of the options should correct the inversion.

// #define TFT_INVERSION_ON
// #define TFT_INVERSION_OFF 

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
#define TFT_CS   17  // Chip select control pin
#define TFT_DC   16  // Data Command control pin
//#define TFT_RST  15  // Reset pin (could connect to NodeMCU RST, see next line)
#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V
#define TOUCH_CS -1

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

//#define SPI_FREQUENCY  20000000
#define SPI_FREQUENCY  27000000
//#define SPI_FREQUENCY  40000000
//#define SPI_FREQUENCY  50000000
//#define SPI_FREQUENCY  60000000

#define SPI_TOUCH_FREQUENCY  2500000
// #define SUPPORT_TRANSACTIONS
`

Bodmer, 2022-07-12:
This is because the latest IDF throws an error if a pin number -1 is used. The message can be ignored.

Read about the problem with the IDF here:
https://github.com/espressif/arduino-esp32/issues/6737

closed 2022-07-12 by Bodmer

JimDrewGH, 2022-07-12:
I can't simply ignore the message because I can't have *any* unwanted serial data going out on the serial port - that is restricted to actual packet data.  Is possible that you change the library so that when something is not defined that it is equal to -1?

Bodmer, 2022-07-12:
In that case simply comment out the setup lines thus:
```
//#define TFT_RST  -1
//#define TOUCH_CS -1
```

JimDrewGH, 2022-07-13:
Perfect!  Thanks!

openstep, 2023-07-23:
HI, 

I am new to arduino and ESP32. I have the same error message when the code tries to set the level on a GPIO port.
Result here is that the relay will not get turned on part of an irrigation system.
This is connected to an 8 channel relay board and it goes in a loop switching realys 0-1-2-3. The first iteration is ok, but the 2nd one I got the error: `E (116335) gpio: gpio_set_level(226): GPIO output gpio_num error`

TXyy2023, 2024-01-08:
I use a simple to solve this problem，we have already know that E (116335) gpio: gpio_set_level(226): GPIO output gpio_num error is usb wrong GPIO，because in propertie it has initialize CS as -1，So I delete all about cs，then it works perfect