Bodmer/TFT_eSPI #1910: ST7735S-G6, ESP8266, eSprite.fillRect wrong color

a442509097, 2022-07-15:
```
void setup() {
  eSprite.setColorDepth(8);
  eSprite.createSprite(128, 160);
  eSprite.fillSprite(TFT_BLACK);
  eSprite.setFreeFont(FMB18);
  eSprite.setTextColor(TFT_WHITE);
}

void loop() {
    eSprite.fillSprite(TFT_BLACK);
    eSprite.fillRect(0, 0, 2, lineHeight, TFT_YELLOW);
    eSprite.fillRect(0, screenHeight - lineHeight, 2, lineHeight, TFT_YELLOW);
    eSprite.pushSprite(0, 0);
}

```
if w<=2 it will show white color

Bodmer, 2022-07-17:
That sketch is incomplete. I have patched together an example and it is working fine:
```
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite eSprite = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object

void setup() {
  tft.init();
  eSprite.setColorDepth(8);
  eSprite.createSprite(128, 160);
  eSprite.fillSprite(TFT_BLACK);
}

void loop() {
  static uint16_t lineHeight  = 40;
  eSprite.fillSprite(TFT_BLACK);
  eSprite.fillRect(0, 0, 2, lineHeight, TFT_YELLOW);
  eSprite.fillRect(0, 160 - lineHeight, 1, lineHeight, TFT_YELLOW);
  eSprite.pushSprite(0, 0);
}
```
Without further information I am not able to help.

closed 2022-07-17 by Bodmer

a442509097, 2022-07-19:
Sorry for not describing it clearly before, but the problem is still there
```
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite eSprite = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object

void setup() {
  tft.init();
  eSprite.setColorDepth(8);
  eSprite.createSprite(128, 160);
  eSprite.fillSprite(TFT_BLACK);
}

void loop() {
  static uint16_t lineHeight  = 40;
  eSprite.fillSprite(TFT_BLACK);
  eSprite.fillRect(0, 0, 2, lineHeight, TFT_YELLOW);
  eSprite.fillRect(0, 160 - lineHeight, 1, lineHeight, TFT_YELLOW);

  eSprite.fillRect(0, 50, 3, lineHeight, TFT_YELLOW); //if>=3,correct color
 
  eSprite.pushSprite(0, 0);

}
```
[image]

aord, 2022-07-20:
I ran your code (2nd code) on a Pico Pi with a ILI9341 240x320 screen, and it worked perfectly. Tested with 1 to 240 pixels wide, gave all yellow bars as expected.

Is your TFT_eSPI setup file configured correctly? 
Which setup are you using?
Do you have another screen to test with?