Bodmer/TFT_eSPI #1914: RP2040 cannot light ST7789

justp7, 2022-07-19:
Hello, I often use TFT_eSP as my screen driver. But I found a problem like this today,TFT_eSPI can't drive my screen.The following is the specific content:
1.I use ST7789 screen and use RP2040 to drive, the screen does not light up after burning the program.The problem arises in
```
  // Toggle RST low to reset
#ifdef TFT_RST
  #if !defined(RP2040_PIO_INTERFACE)
    // Set to output once again in case MISO is used for TFT_RST
    pinMode(TFT_RST, OUTPUT);
  #endif
  if (TFT_RST >= 0) {
    //writecommand(0xff);
    digitalWrite(TFT_RST, HIGH);
    delay(5);
    digitalWrite(TFT_RST, LOW);
    delay(20);
    digitalWrite(TFT_RST, HIGH);
  }
  else writecommand(TFT_SWRST); // Software reset
#else
  writecommand(TFT_SWRST); // Software reset
#endif
```
When I uncomment `writecommand(0xff);` program is available
2. PlatformIO IDE 
3. Board package version is 2.4.71
4. RP2040
5. TFT driver ST7789 
6. SPI（No CS and MISO）

I would be grateful if you could solve the problem!

Bodmer, 2022-07-19:
Thanks for reporting this. I think this is the same problem as #1758.

By default the library uses SPI mode 3 for that display. Can you force mode 0 and try it. It is possible to set the SPI mode by adding this line in the setup file:

#define TFT_SPI_MODE SPI_MODE0

justp7, 2022-07-19:
I tried adding `#define TFT_SPI_MODE SPI_MODE0` in the config file and commenting  `writecommand(0xff);` , but this still doesn't make the ST7789 work.
I don't know what is causing this.

Bodmer, 2022-07-19:
When you say "When I uncomment writecommand(0xff); program is available" does that mean it works OK when it is uncommented?

justp7, 2022-07-19:
Uncommenting `writecommand(0xff);` does make the program usable, but I don't know if this is an unfixed bug.It is indeed available, I don't know who fixed this problem, `writecommand(0xff);` is copied on version 2.4.30, I heard that someone changed this library so that RP2040 can drive ST7789 without CS and MISO