Bodmer/TFT_eSPI #1915: Smooth font - top line masked off - sinkholes on digits

Frank-Bemelman, 2022-07-19:
A phenomena that is avoidable by workaround, but still something I think is not right. 

Situation, smoothfont, 26 pixel height, rotated display, TL_DATUM.

When printing text with y-offset 0, it seems that the very line of the top is missing. See picture. Another effect, wihich is even stranger, that when printing digits, the background under that digit is also wiped to background color, way below the character limit itself.

I can fix it by using an additional offset of 1 for the y coordinate. And the sinkhole can be fixed by setting a tight viewport.

I am using  tft.setRotation(1);  tft. setViewport with TL_DATUM and tft.drawstring for printing the text. 

Here is a small demonstration of both effects:

[image]

If you take a good look, you see that the top of the 0 is sliced off by one pixel. The sinkhole would go all the way to the bottom, but is already masked off somewhat by the active viewport settings. I tried to find a clue in TFT_eSPI.cpp but it a mystery why this only happens with digits and not with alphas. 

BTW, your font tool script Create_font.pde is brilliant. Created a font from my PC arial_narrow, converted it to hex data, included the .h file and voila. At first it is a bit of work and struggle. I felt like Alice in Wonderland - but I can highly recommend everybody to dig in, it is worth the efforts. Have all the extended ascii characters now,as you can also see in the picture. Yeeh!!

Bodmer, 2022-07-19:
This looks like the same problem as #1812 which I thought I had fixed but I see I did not update the Sprite.cpp code.

The latest Github master has been fixed though. The fillHeight varaiable should be signed:

https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L2501-L2502

Let me know if that fixes it for you.

Frank-Bemelman, 2022-07-19:
Thanks a lot, that indeed made all the difference. Still have a line missing on top of the 0, with this arial_narrow font I made. I am using TC_DATUM and when using tft.Drawstring with y-offset zero, the text seem to be positioned 1 pixel too high, shaving off 1 line on top of the 0. Ofcourse that can be circumvented using an y value of 1, but it that is perhaps not intuitive. 

I took the liberty to change this in TDT_eSPI.ccp, around line 4965:
      case TC_DATUM:
        poX -= cwidth/2;
        poY += 1; // add one here to avoid shaving off top line on the tallest glyphs.
        padding += 1;

Suppose that should also apply to TR_DATUM and perhaps TL_DATUM, but the latter is left out of the switch since it is define 0.
....assuming that this is really an issue and not some silly conclusion I made. 

Thanks again, keep up the good work ;-)

Cheers, 
Frank

Bodmer, 2022-07-19:
I think this is due to the ascent value in the font file not being the maximum ascent of any character. The only way around this is to scan every font character but this approach is also flawed (and disabled in the library) because I found rogue characters in some fonts that have a very large ascent value. See here:

https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Smooth_font.cpp#L207-L222

closed 2022-07-19 by Bodmer

Frank-Bemelman, 2022-07-19:
Well, that is interesting. I would think checking between 0x20 and 0x7f would be reasonably safe, but your experience tells different. I might give it a shot, that scanning alternative - just for fun. Learning a lot about fonts and character sets the last days. A week ago I didn't even know what UTF-8 was. Imagine me counting characters the old school way and finding all these funny things.  And I thought I knew a thing or two ;-) Thanks for the explanation!

Frank-Bemelman, 2022-07-21:
Played with that flawed approach and indeed you are spot on. I also had some stray pixels left behind when printing digits, in particular with digit '4'  which seems the tallest guy in the Arial narrow font. Goes higher than the top of "d".  

I uncommented that checking of all characters again and changed the  "top of d" value in the font data manually - that seems a better place to fix it. I am not going to say that the vlw font converter should have done that, but thats just a wild assumption made in lack of knowledge. 

Anyway, stray pixels gone, no unexpected top lines being shaved off, it's all wrapped now in a satisfactionary way.