Bodmer/TFT_eSPI #1916: (2) ST7735S-G6, ESP8266, eSprite.fillRect wrong color

a442509097, 2022-07-20:
Sorry for not describing it clearly before, but the problem is still there
```
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite eSprite = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object

void setup() {
  tft.init();
  eSprite.setColorDepth(8);
  eSprite.createSprite(128, 160);
  eSprite.fillSprite(TFT_BLACK);
}

void loop() {
  static uint16_t lineHeight  = 40;
  eSprite.fillSprite(TFT_BLACK);
  eSprite.fillRect(0, 0, 2, lineHeight, TFT_YELLOW);
  eSprite.fillRect(0, 160 - lineHeight, 1, lineHeight, TFT_YELLOW);

  eSprite.fillRect(0, 50, 3, lineHeight, TFT_YELLOW); //if>=3,correct color
 
  eSprite.pushSprite(0, 0);

}
```
[image]

_Originally posted by @a442509097 in https://github.com/Bodmer/TFT_eSPI/issues/1910#issuecomment-1188612561_

Bodmer, 2022-07-21:
Try reducing the SPI clock speed.

Bodmer, 2022-07-28:
I do not know why this is happening unfortunately. It may be specific to that display+ESP8266 combination as the sketch runs fine on ESP32 and RP2040 processors.

Run this sketch and check all sides of the rectangle are visible.
```
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

void setup() {
  tft.begin();
  tft.setRotation(0);
  tft.fillScreen(TFT_BLACK);
  tft.drawRect(0, 0, tft.width(), tft.height(), TFT_GREEN);
}

void loop() {

}
```

a442509097, 2022-07-29:
```
#define SPI_FREQUENCY  80000000
```
Thanks! This works for me.
[image]

a442509097, 2022-07-29:
If just use TFT_eSPI tft = TFT_eSPI(); , the default #define SPI_FREQUENCY 27000000 will work fine.
[image]

closed 2022-07-29 by Bodmer