Bodmer/TFT_eSPI #1920: Vertical touch scrolling

zuhairmst, 2022-07-21:
I am using ESP32-S3-Devkit-C1 with an LCD having panel controller IC ILI9488. I have elements of UI which are 16 bit .c headers and using pushImage to display them on screen. I want to make setting screen which is larger than my LCD screen so there is a need of vertical scrolling up and down using touch swipe. How is this possible?
Thanks in advance

Bodmer, 2022-07-21:
You can either move a pointer in the image, or plot the image at a partially off screen coordinate and the library will clip the image to the screen area.

The same approach can be used with view ports to scroll in part of the screen area.

closed 2022-07-21 by Bodmer

zuhairmst, 2022-07-21:
Thanks for the response. There are multiple images not one in the settings screen.  And how to scroll using touch?. Maybe I didn't get what you are trying to say. Please elaborate.

Bodmer, 2022-07-21:
You have to redraw the image in a new higher or lower position.

Alternatively you can move the display datum with the viewport functions and then redraw the whole image set to screen.

zuhairmst, 2022-07-22:
Ok. And how to detect touch swipe?