Bodmer/TFT_eSPI #1921: RP2040 to RM68140

tbillion, 2022-07-21:
To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur

i am attempting to combine two files (setup104.h for the rp2040 elements, and setup19.h for the rm68140.h elements) in order to create a custom setup file. i am trying to connect the open smart RM68140 tft to RP2040 and later RP2040W if they indeed ever come off back order. I cant really decipher how the pins for this should be best connected based on the two files that i have chosen as examples. it appears that there are constraints on the rp 2040 pis that an be used, but i cant deciper if the control pin markup used is for the st7735 that the original example comes with or if it is rp2040 specific. guidance is appreciated .

3. IDE (e.g. Arduino or PlatformIO)
Arduino

5. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
name=TFT_eSPI
version=2.4.72
author=Bodmer
maintainer=Bodmer
sentence=TFT graphics library for Arduino processors with performance optimisation for RP2040, STM32, ESP8266 and ESP32
paragraph=Supports TFT displays using drivers (ILI9341 etc) that operate with hardware SPI or 8 bit parallel.
category=Display
url=https://github.com/Bodmer/TFT_eSPI
architectures=*
includes=TFT_eSPI.h

7. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
https://github.com/earlephilhower/arduino-pico
9. Procesor, e.g RP2040, ESP32 S3 etc
rp2040
11. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
RM68140
12. Interface type (SPI or parallel)
parallel
Plus further information as appropriate to the problem:
1. TFT to processor connections used
undefined currently however if i had a preference i like all my ducks in a single line.

```
// This setup is for the RP2040 processor only when used with 8 bit parallel displays
// See SetupX_Template.h for all options available
#define USER_SETUP_ID 787

////////////////////////////////////////////////////////////////////////////////////////////
// Interface
////////////////////////////////////////////////////////////////////////////////////////////
#define TFT_PARALLEL_8_BIT

////////////////////////////////////////////////////////////////////////////////////////////
// Display driver type
////////////////////////////////////////////////////////////////////////////////////////////
#define RM68140_DRIVER

////////////////////////////////////////////////////////////////////////////////////////////
// RP2040 pins used
////////////////////////////////////////////////////////////////////////////////////////////

this is both pin configs from both files but i am not sure which is more correct or if either is correct at all. 

// ESP32 pins used for UNO format board
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR    22  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    -1

  // PIO requires these to be sequentially increasing - do not change
  #define TFT_D0    6
  #define TFT_D1    7
  #define TFT_D2    8
  #define TFT_D3    9
  #define TFT_D4   10
  #define TFT_D5   11
  #define TFT_D6   12
  #define TFT_D7   13

//#define TFT_CS   -1  // Do not define, chip select control pin permanently connected to 0V

// These pins can be moved and are controlled directly by the library software
#define TFT_DC   28    // Data Command control pin
#define TFT_RST   2    // Reset pin

//#define TFT_RD   -1  // Do not define, read pin permanently connected to 3V3

// Note: All the following pins are PIO hardware configured and driven
// The pins are hard-coded at the moment and must not be changed here
// Connections MUST use the pins below
  #define TFT_WR   22

  // PIO requires these to be sequentially increasing - do not change
  #define TFT_D0    6
  #define TFT_D1    7
  #define TFT_D2    8
  #define TFT_D3    9
  #define TFT_D4   10
  #define TFT_D5   11
  #define TFT_D6   12
  #define TFT_D7   13
//*/

////////////////////////////////////////////////////////////////////////////////////////////
// Fonts to be available
////////////////////////////////////////////////////////////////////////////////////////////
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

////////////////////////////////////////////////////////////////////////////////////////////

```

Bodmer, 2022-07-22:
You can shoose the control pins as you wish. The data pins must be sequentially increasing pins but the start pin of the sequence can be chosen. So if GPIO 0 is the data bus start pin:

  #define TFT_D0    0
  #define TFT_D1    1
  #define TFT_D2    2
  #define TFT_D3    3
  #define TFT_D4    4
  #define TFT_D5    5
  #define TFT_D6    6
  #define TFT_D7    7

and another example if GPIO 6 is the data bus start pin:

  #define TFT_D0    6
  #define TFT_D1    7
  #define TFT_D2    8
  #define TFT_D3    9
  #define TFT_D4   10
  #define TFT_D5   11
  #define TFT_D6   12
  #define TFT_D7   13

Bodmer, 2022-07-22:
Also note that for the Pico W the following pins cannot be used as they are dedicated to the WiFi chip interface:

 GPIO29 OP/IP wireless SPI CLK/ADC mode (ADC3) to measure VSYS/3
 GPIO25 OP wireless SPI CS - when high also enables GPIO29 ADC pin to read VSYS
 GPIO24 OP/IP wireless SPI data/IRQ
 GPIO23 OP wireless power on signal

tbillion, 2022-07-23:
THANK YOU!!!

just in case anyone wants the answer completed config
```
// This setup is for the RP2040 processor only when used with 8 bit parallel displays
// See SetupX_Template.h for all options available
#define USER_SETUP_ID 787

////////////////////////////////////////////////////////////////////////////////////////////
// Interface
////////////////////////////////////////////////////////////////////////////////////////////
#define TFT_PARALLEL_8_BIT

////////////////////////////////////////////////////////////////////////////////////////////
// Display driver type
////////////////////////////////////////////////////////////////////////////////////////////
#define RM68140_DRIVER
//#define ILI9341_DRIVER
////////////////////////////////////////////////////////////////////////////////////////////
// RP2040 pins used
////////////////////////////////////////////////////////////////////////////////////////////

// ESP32 pins used for UNO format board

#define TFT_RST  10  // Reset pin
#define TFT_RD    11
#define TFT_WR    12  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_DC   13  // Data Command control pin - must use a pin in the range 0-31
#define TFT_CS   14  // Chip select control pin

  // PIO requires these to be sequentially increasing - do not change
  #define TFT_D0    2
  #define TFT_D1    3
  #define TFT_D2    4
  #define TFT_D3    5
  #define TFT_D4   6
  #define TFT_D5   7
  #define TFT_D6   8
  #define TFT_D7   9

////////////////////////////////////////////////////////////////////////////////////////////
// Fonts to be available
////////////////////////////////////////////////////////////////////////////////////////////
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

////////////////////////////////////////////////////////////////////////////////////////////
```
yields results:

```
@100mhz
TFT_eSPI library test!
Benchmark                Time (microseconds)
Screen fill              61495
Text                     6769
Lines                    54560
Horiz/Vert Lines         5316
Rectangles (outline)     3073
Rectangles (filled)      148568
Circles (filled)         21959
Circles (outline)        21534
Triangles (outline)      13783
Triangles (filled)       52715
Rounded rects (outline)  8711
Rounded rects (filled)   151263
Done!

@133mhz
TFT_eSPI library test!
Benchmark                Time (microseconds)
Screen fill              46244
Text                     5062
Lines                    40897
Horiz/Vert Lines         4004
Rectangles (outline)     2316
Rectangles (filled)      111682
Circles (filled)         16453
Circles (outline)        16156
Triangles (outline)      10331
Triangles (filled)       39563
Rounded rects (outline)  6526
Rounded rects (filled)   113666
Done!

```