Bodmer/TFT_eSPI #1923: ST7355 Driver Issue

HoopmanJ20, 2022-07-22:
Hi I am having an issue and I am looking for some help. I'm in the process of making a pair of ESP32 based smartglasses and I need to get my ST7355 to mirror text to make it readable through my prism optics. I have gone into the drivers for the display and according to the datasheet for this display I need to change MADCTL from 0x37 to 0x56 (Basically change the binary from 00110110 to 01010110 as the first three digits should set the display orientation). When I make this change to the source code, I am getting the display colored set to gold, but no mirroring of the display is visible. Any ideas on why this isn't working or a better way of achieving what I am attempting? Btw I absolutely love this library and I have used it for so many projects and I really appreciate all the work that has gone into making this available for the the community!
[image]

HoopmanJ20, 2022-07-22:
Update: I figured out from the color test example that you can do tft.writecommand(TFT_MADCTL); and then tft.writedata(xxxx);. I now suspect that I am interpreting the datasheet wrong. I believe that it is the first three digits of binary handling the mirroring, but I will attach a picture if anyone wants to correct me for being wrong 
[image]

Bodmer, 2022-07-22:
The TFT_MADCTL in initialised during init(), the init value will depend on the tab colour option you have set in the setup file.

The TFT_MADCTL bits are also changed in a rotation so that code can be modified or used to determine the MADCTL values for a particular display, for example:

[TFT_MADCTL ](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ST7735_Rotation.h#L43)

The MX and MY (mirror X, mirror Y in portrait orientation)) bits are the ones to try.  The MV (mirror Vertical) changes the refresh direction thus has the same effect as MY.