Bodmer/TFT_eSPI #1928: waveshare pico eavl board

klspstck, 2022-07-23:
Good morning,

I want to use the above mentioned board to play around with.
Waveshare provides a Micropython graphics example which works.
Their "C" example doesn't. I don't see any activity on the TFT_CS pin)

The board uses that serial-parallel conversion with 4040,4094 and 7404

Because I used your library successfully with an ESP32 I wanted to try your library with pico and the display.

I am familiar with your concept of "user_setup.h" files.

I am able to compile your colour test and upload it, but the display stays dark.

Obviously something is missing.

I used the following settings in my file:
*****
#define USER_SETUP_ID 0xFFFFFFFF  

#define RP2040_PIO_SPI

#define ILI9488_DRIVER

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT

#define TFT_SPI_PORT 1

#define TFT_MISO  12
#define TFT_MOSI  11
#define TFT_SCLK  10
#define TFT_CS     9  // Chip select control pin
#define TFT_DC     8  // Data Command control pin
#define TFT_RST   15  // Reset pin (could connect to Arduino RESET pin)

// the pico_eval_board has backlight control

#define TFT_BL   13            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define SPI_FREQUENCY  20000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

*****

I do see activity on the important pins TFT_CS, TFT_DC, TFT_SCLK and TFT_MOSI.
Commands are 8 bit and data is 16 bit according to my scope.

Any ideas what I may have forgotten?

Do you need more information or logs?

Kind regards
Klaus P. Stock

Bodmer, 2022-07-23:
Comment out or delete the line:
#define RP2040_PIO_SPI

Bodmer, 2022-07-23:
For that interface type you also need to add this line to the setup:
#define RPI_DISPLAY_TYPE

klspstck, 2022-07-24:
> Links:

Good morning Mr. Bodmer,

I applied both modifications and now I do see the colour demo. :-))

Thank you for your immediate help. This saved my weekend.

I am attaching my config file, which you may use as an additional config 
file in your library.

ON to the next problem now ;-))

Kind regards
Klaus P. Stock

-- 
Klaus P. Stock
Stock & Partner
Advanced Education GmbH

// #warning " ***** waveshare pico eval board used ***** "
// just to see if thefile isincluded
//                            USER DEFINED SETTINGS
//   waveshare pico eval board
//
//   #define RP2040_PIO_SPI   // Note: TFT_eSPI touch functions not available when PIO used.
//
//   #define TFT_SPI_PORT 1
//
//   #define TFT_PARALLEL_16_BIT // **** 16 bit parallel ONLY with RP2040 processor ****
//
//   #define TFT_MISO  12
//   #define TFT_MOSI  11
//   #define TFT_SCLK  10
//   #define TFT_CS     9  // Chip select control pin
//   #define TFT_DC     8  // Data Command control pin
//   #define TFT_RST   15  // Reset pin (could connect to Arduino RESET pin)
//
//   #define ILI9488_DRIVER
//
//   #define TFT_BL   13            // LED back-light control pin
//   #define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)
//
//   #define SPI_FREQUENCY  20000000

#define USER_SETUP_ID 0xFFFFFFFF

// Define to disable all #warnings in library (can be put in User_Setup_Select.h)
#define DISABLE_ALL_LIBRARY_WARNINGS

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

#define RPI_DISPLAY_TYPE

//#define RP2040_PIO_SPI

// Display type -  only define if RPi display

#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)

// #define TFT_INVERSION_ON
// #define TFT_INVERSION_OFF

// ##################################################################################
//
// Section 2. Define the pins that are used to interface with the display here
//
[... 59 lines cut ...]
// #define SPI_FREQUENCY   5000000
// #define SPI_FREQUENCY  10000000
#define SPI_FREQUENCY  20000000
// #define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  55000000 // STM32 SPI1 only (SPI2 maximum is 27MHz)
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

klspstck, 2022-07-27:
Good morning,

one more thing: I also had to turn on inverse colours.

But now everything works as expected.

Kind regards
Klaus P. Stock