Bodmer/TFT_eSPI #1929: TFT_eSPI  XPT2046

rrazmer, 2022-07-23:
To minimize effort to resolve issues the following should be provided as a minimum:
1. After calibration of touch "Touch_calibrate.ino" the coordinates don't match the pressing position.
2. IDE  Arduino 1.8.5
3. TFT_eSPI 2.4.72
4. NA
5. ESP32-WROOM-32 
6. TFT driver ILI9341    2.8 TFT SPI 240x320 V1.2
7. Interface type SPI

(https://github.com/PaulStoffregen/XPT2046_Touchscreen    working good)

Plus further information as appropriate to the problem:
1. TFT to processor connections used
#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)

#define TFT_BL   32            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5  // Chip select control pin
#define TFT_DC    4  // Data Command control pin
#define TFT_RST   22  // Reset pin (could connect to RST pin)

#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.

#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
3.  https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Touch_calibrate/Touch_calibrate.ino

uint16_t calData[5] = { 343, 3834, 323, 3900, 7 };   not working

Bodmer, 2022-07-23:
Try changing the rotation in setup and see if any other rotations do not work.

The calibration values are only valid for the same rotation as was set for the calibration  so if the screen is calibrated in rotation 3 then the values are only valid for that rotation in future sketches.

rrazmer, 2022-07-24:
https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Touch_calibrate/Touch_calibrate.ino

this sketch is not working - without playing with rotation.
After calibration the values are copied to sketch and touch_calibrate();   - commented.
  uint16_t calData[5] = { 286, 3534, 283, 3600, 6 };   tft.setTouch(calData);  - uncommented
The touch is not working properly.
Tried on 3 screens ILI9341 2.8 TFT SPI 240x320 V1.2

and https://github.com/PaulStoffregen/XPT2046_Touchscreen    working good.
Regards

Bodmer, 2022-07-28:
Run the "Test_Touch_Controller" sketch ad tell me what the x and y values are for each of the corners of the display. This may help me to diagnose the problem.

Bodmer, 2022-08-03:
Assumed resolved.

closed 2022-08-03 by Bodmer

summerfind, 2023-06-23:
> https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Touch_calibrate/Touch_calibrate.ino
>
> this sketch is not working - without playing with rotation. After calibration the values are [...]

Hi [rrazmer](https://github.com/rrazmer)
I tested the example in XPT2046 of your link, and just got output like below when touched ILI9341:
.........................
Pressure = 4095, x = 0, y = 0
Pressure = 4095, x = 0, y = 0
Pressure = 4101, x = 26, y = 18
Pressure = 4101, x = 26, y = 18
Pressure = 4095, x = 0, y = 0
Pressure = 4095, x = 0, y = 0
 .................................

what's wrong about?
Thanks.

rrazmer, 2023-06-23:
Now is ok