Bodmer/TFT_eSPI #1931: 3.5' TFT ILI9486(Parallel 8bit) with ESP32, Touch Control?

RoboticsAayush, 2022-07-24:
Hi, I was working with a 3.5' TFT display Driver :ILI9486(Parallel 8bit) and ESP32, and thanks to the library I was able to full control the display, but I am unable to get the touch functionality working with the same.
I tried adding #define TOUCH_CS 22 to my code but the display goes completely white.
Could anyone guide me?

Here is a link to the display  : https://robu.in/product/3-5-inch-ili9486-tft-touch-shield-lcd-module-480x320-for-arduino-uno/
Here is the ESP32 : https://robu.in/product/esp-wroom-32-wifi-bluetooth-networking-smart-component-development-board/
@Bodmer 
Thanks
Aayush

Bodmer, 2022-07-24:
The display you have has a 4 wire analogue resistance interface. You can use the library and wiring described here.:
https://github.com/s60sc/Adafruit_TouchScreen

closed 2022-07-24 by Bodmer

dipankarpanda, 2022-09-18:
Hi @RoboticsAayush have you got it working?