Bodmer/TFT_eSPI #1932: [28][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected

gjssss, 2022-07-25:
Hello, I'm learning ESP32 and when I'm using TFT_eSPI, I got a problem.
I have a `ESP32 DEVKITV1` and a `1.44 128*128 TFT` with `ST7735`
I selected the `Setup47` in `User_Setup_Select.h`
```
#include <User_Setups/Setup47_ST7735.h>            // Setup file for ESP32 configured for ST7735 128 x 128 animated eyes
```
Here is my pins setting
```
// Generic ESP32 setup
#define TFT_MISO -1
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   -1 // Not defined here, chip select is managed by sketch
#define TFT_DC    2
#define TFT_RST   4  // Connect reset to ensure display initialises
```
I have read the Issues[#1767](https://github.com/Bodmer/TFT_eSPI/issues/1767), and I have tried to change `MISO` and `CS` to 40 or -1, but It is not work.
Perheps it is my wrong config or operation that make the problem, because I am just contacted ESP32 for two weeks.

By the way, There is a strange situation. When I touch some pins(I dont know exactly but just touch ESP32 pins), I can work.
🥺

gjssss, 2022-07-27:
https://user-images.githubusercontent.com/74909529/181293214-5a254b74-83b1-478a-8489-5bda8c4db06b.mp4

I set cs to pin_15, there is no error any more. I don't know what happened, but when I touch it, it can light.

Is there something wrong with my screen or board?

Bodmer, 2022-07-28:
Start with defining and connectingthe chip select pin as otherwise none of the examples will work.

Run the Read_User_Setup sketch and check the compiler is picking up the right setup file ant that the output is as you expect.

gjssss, 2022-07-29:
Thank you.
It worked when I defined the CS and it stopped flashing as I set the backlight to 3.3V.   : )

closed 2022-07-29 by gjssss