Bodmer/TFT_eSPI #1933: Compatibility 8Bit 8080-8 parallel touch screen ST7735S

podaen, 2022-07-27:
Hi, I am intressted to use you library for my project. Soo I was wondering if could make this tiny touch screen working with this library?

[manual](http://surenoo.tech/download/04_STP/0177/STP0177G-128160.pdf?spm=a2g0o.detail.1000023.23.1138285c2dvUjo&file=STP0177G-128160.pdf)

This is what I have for now to connect.
[image]

Thanks in advance!

closed 2022-07-28 by Bodmer

podaen, 2022-07-30:
Is D0-7 the same as DB7 on my shematic?

Bodmer, 2022-07-30:
Yes, Dx = DBx
All 8 data lines must be connected to transfer the data.

podaen, 2022-07-30:
In parallelmode than? one bit each

podaen, 2022-08-02:
I was confiused. This display doesn't have the option to choose between SPI and Parallel. So I need to connect it. But I'm running short on pins. I suspect Y-,X-,Y+,X+ are touchscreen pins inpins.They could be connected to only input pins GPI34,GPI35,GPI39,GPI36 I think.