Bodmer/TFT_eSPI #1948: 8Bit Parallel Support on upper GPIO pins for ESP32-S3

ddimensia, 2022-08-04:
The new LillyGo ESP32-S3 T-Display device uses an 8-bit parallel display with the data pins connected to GPIO 39-48 (43/44 are excluded as they are UART0).  The current ESP32-S3 parallel support only reads from the lower 32 GPIO pins, so it won't support the new device.

Board: LillyGo T-Display-S3
Processor: ESP32-S3
TFT Driver: ST7789
Interface: Parallel

Display/device pinout

[image]

ddimensia, 2022-08-04:
This post indicates that using a dedicated gpio bundle may get around the slowness added when using GPIO pins above 31. https://www.esp32.com/viewtopic.php?t=27963#p98194

Bodmer, 2022-08-04:
It is possible. If you replace the library header file TFT_eSPI_ESP32_S3.h with this one then the 8 bit bus GPIO can be 32 or higher:
[TFT_eSPI_ESP32_S3.zip](https://github.com/Bodmer/TFT_eSPI/files/9261054/TFT_eSPI_ESP32_S3.zip)

Should work but it is untested, let me know if this works.

Bodmer, 2022-08-04:
Replace the library header TFT_eSPI_ESP32_S3.h with this one and it shoule work:
[TFT_eSPI_ESP32_S3.zip](https://github.com/Bodmer/TFT_eSPI/files/9261110/TFT_eSPI_ESP32_S3.zip)

Let me know if it works.

lbxhacker, 2022-08-08:
i also use ESP32-S3 with ESP-IDF,It's can work PARALLEL_8_BIT ,

ddimensia, 2022-08-08:
I should have the devices in a day or two and will give it a try. I've been playing with a non-parallel display I had floating around in the mean time.

MartinNohr, 2022-09-12:
I have several of these being shipped. I'll test as soon as I get them.

9Rune5, 2022-10-12:
LilyGO released their own patch: https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/b12c08a3a121f8938aa08ca27ae44002d40119a8/lib/TFT_eSPI/micky_commit.patch

9Rune5, 2022-10-15:
> Replace the library header TFT_eSPI_ESP32_S3.h with this one and it shoule work: [TFT_eSPI_ESP32_S3.zip](https://github.com/Bodmer/TFT_eSPI/files/9261110/TFT_eSPI_ESP32_S3.zip)
> 
> Let me know if it works.

I'm still getting acquainted with PlatformIO, so I may have missed something.

It looks like you've removed the definition og GPIO_DIR_MASK:

```
.pio/libdeps/ESP32-S3-DevKitC-1/TFT_eSPI/TFT_eSPI.cpp: In member function 'uint8_t TFT_eSPI::readcommand8(uint8_t, uint8_t)':
.pio/libdeps/ESP32-S3-DevKitC-1/TFT_eSPI/TFT_eSPI.cpp:993:10: error: 'GPIO_DIR_MASK' was not declared in this scope
   busDir(GPIO_DIR_MASK, INPUT);
```

readByte() is also struggling (one warning per TFT_Dx):
```
.pio/libdeps/ESP32-S3-DevKitC-1/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c: In member function 'uint8_t TFT_eSPI::readByte()':
<command-line>: warning: right shift count >= width of type [-Wshift-count-overflow]
.pio/libdeps/ESP32-S3-DevKitC-1/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:106:16: note: in expansion of macro 'TFT_D0'
   b  = (((reg>>TFT_D0)&1) << 0);
                ^~~~~~
```

My defines:
```
              -DBOARD_HAS_PSRAM
              -DCONFIG_IDF_TARGET_ESP32S3
              -DUSER_SETUP_LOADED
              -DST7789_DRIVER=1
              -DTFT_BL=38
              -DTFT_CS=6
              -DTFT_DC=7
              -DTFT_RST=5
              -DTFT_WR=8
              -DTFT_RD=9
              -DTFT_D0=39
              -DTFT_D1=40
              -DTFT_D2=41
              -DTFT_D3=42
              -DTFT_D4=45
              -DTFT_D5=46
              -DTFT_D6=47
              -DTFT_D7=48
              -DTFT_PARALLEL_8_BIT
              -DTFT_WIDTH=170
              -DTFT_HEIGHT=320
              -DTFT_INVERSION_ON

```

nicpottier, 2022-10-20:
Hi guys, just got this board and trying to get the TFT going in platformio. I target multiple boards so would love if I can stick to the mainline of tft_espi. Anything I can do to help?

nicpottier, 2022-10-20:
I can make it work with the lilygo supplied fork they have here FWIW:
  https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/lib/TFT_eSPI

ddimensia, 2022-10-20:
I've verified the LilyGo fork works as intended.

closed 2022-10-20 by ddimensia

9Rune5, 2022-10-20:
Yes, the lilygo fork works, but it would be nice to have this merged into the main library.

h3ndrik, 2022-11-04:
please reopen this. it hasn't been completed.