Bodmer/TFT_eSPI #1962: cannot read pixel rm68140

tbillion, 2022-08-15:
To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
tft.readPixel does not work with RM68140, not sure if i missed the documentation on this function not being part of the driver which is possible. i have a ton of these screens though, and if this doesn't function its going to spoil it a little bit for me. 

3. IDE (e.g. Arduino or PlatformIO)
arduino 
5. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
lib from github
7. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
latest earl philhower 2.3.2
9. Procesor, e.g RP2040, ESP32 S3 etc
10. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
RM68140
11. Interface type (SPI or parallel)
parallel

```
for (i = hh - 1; i >= 0; i--) {
    for (j = 0; j < ww; j++) {

      // if you are attempting to convert this library to use another display library, this
      // is where you may run into issues
      // the libries must have a readPixel function
      rgb = tft.readPixel(j, i);
      tft.setTextColor(TFT_GREEN , TFT_BLACK);
      tft.drawString(String(tft.readPixel(j, i)), 200, 200);
    }
  }
```

i simplified the code i am trying to achieve and dumped it out of my code into an example and the result is always the same the variable is 0.

Bodmer, 2022-08-15:
Parallel but reads are currently not supported on RP2040. I will add as a TODO for the future.

Using sprites mat be a workaround option as pixel read works in sprites.

tbillion, 2022-08-15:
does this work on parallel?

void     readRect(int32_t x, int32_t y, int32_t w, int32_t h, uint16_t *data);

Bodmer, 2022-08-15:
Typo corrected:
Parallel bus reads are not supported on RP2040.

tbillion, 2022-08-15:
awe bummer , thanks bodmer for all the great and amazing things you do! the community wouldn't be as much fun without you !