Bodmer/TFT_eSPI #1970: STM32F4xx SPI 3 DMA hanging

jfkwhat, 2022-08-19:
IDE: PlatformIO
TFT_eSPI Library Version: 2.4.75
Processor: STM32F411RE (Nucleo board)
TFT driver: ILI9341 (https://www.adafruit.com/product/1770)
Interface type: SPI
Problem: DMA was not working with SPI 3.  SPI3 works fine without DMA.  I think the interrupt for stream 5 was not enabled so I made a change to the TFT_eSPI_STM32.c file inside the initDMA function and DMA works correctly.  Could be a set-up issue too so I have attached my set-up file.  Working on a similar project to the bouncy balls DMA example.

Was:

```
  #if (TFT_SPI_PORT == 1)
    HAL_NVIC_EnableIRQ(DMA2_Stream3_IRQn);  // Enable DMA end interrupt handler
  #elif (TFT_SPI_PORT == 2)
    HAL_NVIC_EnableIRQ(DMA1_Stream4_IRQn);  // Enable DMA end interrupt handler
  #endif
'''

Added a check to initialise DMA1_Stream5_IRQn:
```
  #if (TFT_SPI_PORT == 1)
    HAL_NVIC_EnableIRQ(DMA2_Stream3_IRQn);  // Enable DMA end interrupt handler
  #elif (TFT_SPI_PORT == 2)
    HAL_NVIC_EnableIRQ(DMA1_Stream4_IRQn);  // Enable DMA end interrupt handler
  **#elif (TFT_SPI_PORT == 3)
    HAL_NVIC_EnableIRQ(DMA1_Stream5_IRQn);**
  #endif
```

[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/9385790/User_Setup.zip)

Bodmer, 2022-08-23:
OK, I will add at the next update. Thanks.

Bodmer, 2023-02-04:
Update added, thanks.

closed 2023-02-04 by Bodmer