Bodmer/TFT_eSPI #1983: The result of TFT_eSPI_Button#justPressed() and #justReleased() is indeterminate on startup.

william-ferguson-au, 2022-08-25:
The result of TFT_eSPI_Button#justPressed() and #justReleased() is indeterminate on startup.

Because TFT_eSPI_Button doesn't specify the value of laststate and currstate during construction the result of #justPressed() and #justReleased() is indeterminate on startup. and will depend on previous memory assignment.

This means that following the suggested loop paradigm of  marking any pressed button as pressed, and all other buttons as not pressed before checking #justPressed() and #justReleased() will almost certainly mean that #justReleased==true for all buttons on the first iteration as it is more likely that currstate will have some value other than zero/false.

Eg
```
    uint16_t t_x = 0, t_y = 0; // To store the touch coordinates
    bool pressed = tft->getTouch(&t_x, &t_y);
    if (pressed) {
        ESP_LOGD(TAG, "Screen pressed x=%d y=%d", t_x, t_y);
    }
    for (uint8_t button_id = 0; button_id < END_SCREEN_BUTTONS; button_id++) {
        const button_t* button = buttons[button_id];
        if (button == nullptr) {
            continue;
        }
        TFT_eSPI_Button* tft_button = button->tft_button;
        if (pressed && tft_button->contains(t_x, t_y)) {
            ESP_LOGD(TAG, "Button=%d pressed=%d", button_id, true);
            tft_button->press(true);  // tell the button it is pressed
        } else {
            // Mark all buttons that are not the only currently being pressed as not pressed.
            // This is required so that the #justPressed/#justReleased method work.
            tft_button->press(false);  // tell the button it is not pressed
        }
    }

    // For each button,
    //   if it has just been pressed then draw it as inverted and call its pressed_function.
    //   if it has just be released then draw it as normal and call its release_function.
    for (uint8_t button_id = 0; button_id < END_SCREEN_BUTTONS; button_id++) {
        const button_t* button = buttons[button_id];
        if (button == nullptr) {
            continue;
        }
        TFT_eSPI_Button* tft_button = button->tft_button;
        if (tft_button->justPressed()) {
            ESP_LOGI(TAG, "Button=%d just pressed", button_id);
            tft_button->drawButton(true, button->label);
            if (button->touched != nullptr) {
                ESP_LOGI(TAG, "Calling pressed func for button=%d", button_id);
                button->touched();
            }
        } else if (tft_button->justReleased()){
            ESP_LOGI(TAG, "Button=%d just released", button_id);
            tft_button->drawButton(false, button->label);
            if (button->released != nullptr) {
                ESP_LOGI(TAG, "Calling released func for button=%d", button_id);
                button->released();
            }
        }
    }
```

IDE: CLion
TFT_eSPI 
Processor: ESP32
TFT Driver: NA (ILI9486) 
Interface: SPI

Pull request incoming

william-ferguson-au, 2022-08-25:
Pull request at https://github.com/Bodmer/TFT_eSPI/pull/1984

Bodmer, 2022-08-25:
Thanks for reporting this, now fixed.

commit afd72efb (2022-08-25): Fix #1983
closed 2022-08-25 by Bodmer