Bodmer/TFT_eSPI #1990: Different versions of ILI9341 tft screen

shalom198, 2022-08-28:
[image]

The bottom screen working properly is at this link
https://www.aliexpress.com/item/4000444194938.html?spm=a2g0o.order_list.0.0.21ef1802uoPL5t

The top screen that has a malfunction in its display is at this link
https://www.aliexpress.com/item/1005004542864792.html?spm=a2g0o.order_list.0.0.21ef1802uoPL5t

Why does this happen and how do you fix it?
And on this occasion, thank you for the wonderful driver you created

shalom198, 2022-08-28:
the seler send me this files, but i dont know how to use it

//ILI9341
/***************************************************/

#include<reg51.h>
#include<absacc.h>
#include<intrins.h>
#include<string.h>
#define uchar unsigned char
#define uint unsigned int

//引脚连接定义
sbit cs        =P3^4;//片选
sbit reset     =P3^2;//复位
sbit rs        =P3^3;//地址/数据切换
sbit sda       =P3^1;// SDI  MOSI
sbit scl       =P3^0;//时钟
//SDO引脚可以不连接

uchar bdata bitdata;
sbit bit7=bitdata^7;
sbit bit6=bitdata^6;
sbit bit5=bitdata^5;
sbit bit4=bitdata^4;
sbit bit3=bitdata^3;
sbit bit2=bitdata^2;
sbit bit1=bitdata^1;
sbit bit0=bitdata^0;

void  OTP_Program();
void  delay(uint t);
void  Contrast_Adjust();
uchar code Zk_ASCII8X16[];

unsigned char code gImage_qq[3200];
unsigned char code gImage_ww[3200];

void delay(uint time)
{
 uint i,j;
  for(i=0;i<time;i++)
   for(j=0;j<50;j++);
}

void LCD_CtrlWrite_IC(uchar c)
{
cs=0;

rs=0;
bitdata=c;

sda=bit7;scl=0;scl=1;
sda=bit6;scl=0;scl=1;
sda=bit5;scl=0;scl=1;
sda=bit4;scl=0;scl=1;
sda=bit3;scl=0;scl=1;
sda=bit2;scl=0;scl=1;
sda=bit1;scl=0;scl=1;
sda=bit0;scl=0;scl=1;
cs=1;

}
void  LCD_DataWrite_IC(uchar d)
{
cs=0;

rs=1;
bitdata=d;

[... 667 lines cut ...]
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XBE,0XF7,0XF5,0XCD,0XAD,0XBC,0X69,0XBC,
0X05,0XBC,0X03,0XC4,0X01,0XC4,0X01,0XC4,0X01,0XC4,0X01,0XC4,0XE2,0XC3,0X24,0XC4,
0X68,0XBC,0XAC,0XBC,0XD3,0XCD,0X9E,0XF7,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0X5B,0XEF,0X98,0XDE,0X56,0XD6,0X57,0XD6,0X57,0XD6,0X56,0XD6,0X77,0XDE,0X1B,0XEF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,
};
//单片机存储容量不够，只能定义两个空数组，无法完成图片显示实验
#else
unsigned char code gImage_qq[]={
0,0,
};
unsigned char code gImage_ww[]={
0,0,
};
#endif

pipi61, 2022-08-28:
try check in https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h other driver in line 46
in more detail https://github.com/Bodmer/TFT_eSPI/issues/1172

shalom198, 2022-08-28:
hi thanks, its better but still not good 
here is the good
[image]

ihere is the bad, after change to second driver

[image]

shalom198, 2022-08-29:
i found the right config, just need to opsite
  tft.invertDisplay( true)

closed 2022-08-29 by shalom198