Bodmer/TFT_eSPI #1991: Add SPIClass object parameter for <init> function

StarsRiver-der, 2022-08-29:
A **hal Error** will be caused when I sharing one spi_bus(VSPI) with multi devices (in my case is `microSD` and `GC9A01A`), the reason is, `TFT_eSPI::begin_tft_write` and `end_write` function only use `spi.beginTransaction()` and `spi.endTransaction()` to control the data stream, but not release the spi lock(by using `spi.end()`), 
The Error message is: `[esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x400dea94 arg=0x3ffbdd5c`
1. Procesor: ESP32
2. IDE PlatformIO
3. Interface type: SPI
6. TFT driver: GC9A01A

My temporary solution is: 
```
inline void TFT_eSPI::begin_tft_write(void){
  spi.begin(**);
  // do something
}

inline void TFT_eSPI::end_tft_write(void){
  // do something
  spi.end();
}
```

But i think add more parameter for `init()` could be better, like this `tft.init(CS, SPI_obj, Freq)`. So that I can control SPI bus begin or end manually, the code be like this:
```
SPIClass SPI_obj= SPIClass(VSPI);
tft.init(TFT_CS, SPI_obj, Freq);
microSD.init(SD_CS, SPI_obj, Freq);

SPI_obj.begin(CLK, MISO, MOSI, TFT_CS);
tft.do_something_1();
tft.do_something_2();
tft.do_something_3();
SPI_obj.end();

SPI_obj.begin(CLK, MISO, MOSI, SD_CS);
microSD.do_something();
SPI_obj.end();
```

Bodmer, 2022-08-30:
You can get the SPI class instance by using the library function:
```
static   SPIClass& getSPIinstance(void);
```
Then control begin/end in the sketch.

StarsRiver-der, 2022-08-30:
> You can get the SPI class instance by using the library function:
> 
> ```
> static   SPIClass& getSPIinstance(void);
> ```
> 
> Then control begin/end in the sketch or pass instance to microSD library.

ok, that is enough.

closed 2022-08-31 by StarsRiver-der

baloyou, 2022-09-15:
> 

hello, I have the same question as you, can you share your code? Thank you very much!!

StarsRiver-der, 2022-09-16:
> > 
> 
> hello, I have the same question as you, can you share your code? Thank you very much!!

I code a common class as template for fully comply with different devices.
A `IO_SPI.cpp` is parent class that 'my_device.cpp' needs to extend.
you can see it in my gitlab, please refer to.
[IO_SPI.cpp](http://git.starsriver.net:8110/starsriver/circuits-design/-/blob/master/Module/ESP32_WROVER_EVB/firmware/SmartKnob/src/module/IO_SPI.cpp)
[display class](http://git.starsriver.net:8110/starsriver/circuits-design/-/blob/master/Module/ESP32_WROVER_EVB/firmware/SmartKnob/src/module/dev/display/GC9A01A.cpp)

in `display class`, two methods below are used to control spi. Where `Device` is announced by `TFT_eSPI Device = TFT_eSPI();`
```
void begin_spi() { Device.getSPIinstance().begin(CLK, MISO, MOSI, CS); }
void end_spi() { Device.getSPIinstance().end(); }
```

baloyou, 2022-09-16:
> I code a common class as template for fully comply with different devices. A `IO_SPI.cpp` is parent class that 'my_device.cpp' needs to extend. you can see it in my gitlab, please refer to. [...]

Okay, I'll try, thanks. :)