Bodmer/TFT_eSPI #1993: ESP32, ST7789 and Animated_dial sketch - missing needle #1853

PeterMikeSmooth, 2022-08-31:
Hello Dodmer

Tried Animated_dial sketch as well on ESP-C3-MINI dev kit and needle is missing as well as in issue #1853. I use TFT_eSPI 2.4.72 on arduino 1.8.12. Espressif Systems 2.0.4

Here is my SPI conf : 

```
#define TFT_MISO -1
#define TFT_MOSI 6
#define TFT_SCLK 4
#define TFT_CS 7 // Chip select control pin
#define TFT_DC 18 // Data Command control pin
#define TFT_RST 19 // Reset pin (could connect to RST pin)
```

Half-duplex used :
```
#define TFT_SDA_READ
```

Also I needed to define theses in the User_Setup.h (I don't really understand why):
```
#define VSPIQ_IN_IDX 64
#define VSPIQ_OUT_IDX 64
#define VSPID_IN_IDX 65
#define VSPID_OUT_IDX 65
```

Memory request seems ok according to your suggestions :
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb 7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
Main Sprite allocated!
Needle Sprite allocated!
```

With the TFT_ReadWrite_Test sketch, the ST7789 screen is purple and the output gives :
```
Pixel value written = 200
Pixel value read = 40
ERROR ^^^^
Pixel value written = 400
Pixel value read = 800
ERROR ^^^^
Pixel value written = 800
Pixel value read = 1
ERROR ^^^^
Pixel value written = 1000
Pixel value read = 40
ERROR ^^^^
Pixel value written = 2000
Pixel value read = 800
ERROR ^^^^
Pixel value written = 4000
Pixel value read = 1
ERROR ^^^^
Pixel value written = 8000
Pixel value read = 40
ERROR ^^^^
Pixel value written = 1
Pixel value read = 800
ERROR
```

Can you please advise on this issue ?
Thanks
Peter

PeterMikeSmooth, 2022-08-31:
Thank you fot the very quick reply. 

Tried speed as low as :  
```
#define SPI_FREQUENCY   1000000
```

And
```
#define SPI_READ_FREQUENCY  1000000
```

Display the background take longer time, but still no needle :\

Bodmer, 2022-08-31:
Does the display have a MISO and MOSI pin, if so then you need to connect MISO.

If the display has a bidirectional SDA data pin then currently that is not supported on the C3. The original ESP32 should work fine though.

PeterMikeSmooth, 2022-08-31:
Indeed, the display has no MISO. Just SDA. Ok if not supported by esp32-c3 then  ... 

I tried on original ESP32, but same issue

Bodmer, 2022-09-02:
With the ESP32 it is required that the native VSPI port pins are used to read the display:

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18

closed 2022-09-02 by Bodmer

PeterMikeSmooth, 2022-09-27:
Ok thank you very much Bodmer !

PeterMikeSmooth, 2023-01-12:
> With the ESP32 it is required that the native VSPI port pins are used to read the display:
> 
> #define TFT_MISO 19 #define TFT_MOSI 23 #define TFT_SCLK 18

Does it mean that it can only work with display have a MISO and MOSI pin ? 
If my display has bidirectional SDA then I will not be able to read pixels with an original ESP32 ?

PeterMikeSmooth, 2023-01-28:
Thank you for the example setup, but unfortunately I still can't read : 
``` 
Pixel value written = 8
Pixel value read    = 40
ERROR                 ^^^^
```  

Would you please give any TFT purchasing recommendation to get it work ? I need at least 240*240 resolution. 

Thank you very much