Bodmer/TFT_eSPI #2000: TFT eSPI, Seeed Xiao Rp2040 and GC9A01 not working anymore...

djairjr, 2022-09-07:
To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
2. IDE (e.g. Arduino or PlatformIO)
3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
5. Procesor, e.g RP2040, ESP32 S3 etc
6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
7. Interface type (SPI or parallel)

Plus further information as appropriate to the problem:
1. TFT to processor connections used
2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
4. Screen shot pictures showing the problem  (just drag and drop in message window)

The idea is to provide sufficient information so I can setup the exact same (or sufficiently similar) scenario to investigate and resolve the issue.

Put three ticks ( ` character on key next to "1" key) at the start and end of short pasted code segments to avoid format/markup anomolies and to use a monspaced font. Example:

```
  Serial.begin(115200);
  tft.init();
```

Bodmer, 2022-09-08:
Use the GPIO number only in the setup file.

For example:

#define TFT_MOSI 3    // P3
#define TFT_SCLK 2   // P2
#define TFT_CS 1    // P1
#define TFT_DC 6    // P6
#define TFT_RST 7    // P7

closed 2022-09-08 by Bodmer

djairjr, 2022-09-09:
Does not work. No image at all.

Using XIAO RP2040 SPI Pinout at: [https://wiki.seeedstudio.com/XIAO-RP2040/](url) and earlephilhower core.

`//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

#define RP2040_PIO_SPI // So com Erle 
#define GC9A01_DRIVER // Round TFT SPI Display 

#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // GC9A01 240 x 240

#define TFT_MISO 9

#define TFT_SCLK 8
#define TFT_MOSI 10 
#define TFT_CS   7
#define TFT_DC   4
#define TFT_RST  5

#define TFT_BL   -1            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
#define PIXEL_DOUBLE // Testando

//#define SPI_FREQUENCY  20000000
//#define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000
//#define SPI_FREQUENCY  62500000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  5000000
// #define SPI_TOUCH_FREQUENCY  2500000 - Copy from Setup200_GC9A01
`

djairjr, 2022-09-09:
I will try with slower spi frequency...

djairjr, 2022-09-09:
Bodmer, I'm doing every possible test here. I redid all the cabling and changed my custom configuration file according to your recommendations. I did a check on Earle's core to check the SPI pinout for the Seeed Xiao Rp2040, which is correct. I also changed the User_Setup.h to load the GC9A01 Driver with the indicated pins. Still, I still don't have an image. I have another system already mounted and recorded on a Teensy 3.2 and I am using that system to test the displays (which are new) all four are ok. I also lowered the spi speed to 27 (before it worked with 40) and still, I have no image. In Teensy it is not necessary to turn on the BL pin.

Apparently the driver loads correctly. Using the Read_User_Setup, I have the following messages:

`
TFT_eSPI ver = 2.4.72
Processor    = RP2040
Transactions = Yes
Interface    = SPI
Display driver = 9A01
Display width  = 240
Display height = 240

MOSI    = GPIO 10
MISO    = GPIO 9
SCK     = GPIO 8
TFT_CS   = GPIO 7
TFT_DC   = GPIO 4
TFT_RST  = GPIO 5

TFT_BL           = GPIO -1
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00

`

Bodmer, 2022-09-09:
The pin selection you have now made means that the hardware SPI cannot be used so you must use the PIO SPI.

I have connected a GC9A01 display to a Pico board and created the following setup file and it is working fine:

```
#define RP2040_PIO_SPI

#define GC9A01_DRIVER

#define TFT_MOSI  10
#define TFT_MISO  9
#define TFT_SCLK  8
#define TFT_CS   7  // Chip select control pin
#define TFT_DC   4  // Data Command control pin
#define TFT_RST  5  // Reset pin (could connect to Arduino RESET pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  33000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
```

Note that typically the display does not have a 5V regulator so only connect 3.3V for the power rail, do NOT use 5V.