Bodmer/TFT_eSPI #2018: ILI9488 + ESP32 Dev Module

Sudupatil, 2022-09-14:
I read all documentation and even last discussion on this topic. but still I don't know what i am doing wrong. Can you help me to find it out.

I am using TFT_eSPI-2.4.72 version of your Library.

Display purchased from 
https://probots.co.in/3-5-inch-tft-lcd-with-touch-screen-display-module-spi-interface.html?gclid=EAIaIQobChMI6MOHvtaT-gIVLJlmAh1cuwbnEAYYAyABEgLo7PD_BwE

Here are pin Configuration
SCK		   = GPIO 18
SDI(MOSI)   = GPIO 	23
DC		   = GPIO 4
RESET	   = GPIO 2
CS		   = GPIO 15

T_IRQ	   = GPIO -
T_DO	   = GPIO 19
T_DIN	   = GPIO 23
T_CS	   = GPIO 13
T_CLK	   = GPIO 18

**READ_USER_SETUP.INO** output
[code]
TFT_eSPI ver = 2.4.72
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 4
TFT_RST  = GPIO 2
TOUCH_CS = GPIO 13

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50
[/code]

**Test_Touch_Controller.ino** output
Starting...
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 

Got some readings when Touched 

x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 1452     y: 1789     z: 2087 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 1115     y: 982     z: 2228 
x: 1108     y: 982     z: 2153 
x: 0     y: 0     z: 4095 
x: 1907     y: 1526     z: 2187 
x: 1913     y: 1538     z: 2055 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 3658     z: 26 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 2037     y: 1243     z: 2344 
x: 2038     y: 1236     z: 2299 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 
x: 0     y: 0     z: 4095 

With Example On_Off_Button.ino, i can see the arrow marks but that was very fast before i click those passed to next. 

With Example Touch_Controller_Demo.ino, I am not able to see any thing(white screen). please see the video. Having said that, once I was able to click all 4 corner rectangles.  then I am not able to do that again(even after SPIFFS.remove(CALIBRATION_FILE)).

for some example I am betting white screen, with no change in setting or wiring if I load any other graphic program then that works fine. not sure  why it's happening.

if you need any other information then please let me know.

[Setup21_ILI9488.h.txt](https://github.com/Bodmer/TFT_eSPI/files/9563571/Setup21_ILI9488.h.txt)
[Setup20_ILI9488.h.txt](https://github.com/Bodmer/TFT_eSPI/files/9563573/Setup20_ILI9488.h.txt)
[User_Setup_Select.h.txt](https://github.com/Bodmer/TFT_eSPI/files/9563575/User_Setup_Select.h.txt)
[User_Setup.h.txt](https://github.com/Bodmer/TFT_eSPI/files/9563576/User_Setup.h.txt)
[image]
[image]

https://user-images.githubusercontent.com/99311112/190084404-bc27e539-5cdc-4c82-9df3-d37880f6447a.mp4

Sudupatil, 2022-09-14:
Hello 

Here is link to the video.

https://1drv.ms/u/s!AlcBNDPDdJHVmBhNFjyUCJsw4zKR?e=SHV58Z

Thanks and regards
Sudarshan Patil

Bodmer, 2022-09-14:
Thanks for the complete description. I think this is a bug.

I have added [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Touch.cpp#L110) to cover the case where x and y report as 0. You will need to edit your copy of the library accordingly or download and use the master library zip file from Github.

Let me know if this works.

Sudupatil, 2022-09-15:
Hello,

Thank you for your quick response, but nothing has changed in output. I removed the old library and downloaded new and made necessary changes.
What am I doing wrong? Are those pin configurations okay? or display unit is not working?

No touch input.  With Touch Test X, Y And Z are now always zero(0). previously sometime it's giving me some results but not now. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Sudupatil, 2022-09-15:
Hello,

Thank you for your quick response, In Morining I Removed folder from exporer and download new from ARDUINO IDE. Nothing was working. But at evening I deleted it again and download it from https://github.com/Bodmer/TFT_eSPI/ and add from Menu ZIP library. and after pin configuration, Button on off started working.

But i tried Keyboard example and that isn't working. can you help me with that.

Thanks and regards
Sudarshan Patil

Read_User_Setup.ino Output

[code]
TFT_eSPI ver = 2.4.72
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 4
TFT_RST  = GPIO 2
TOUCH_CS = GPIO 13

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
Touch SPI frequency   = 2.50
[/code]

Test_Touch_Controller.ino Output
Starting...
x: 0     y: 3719     z: 21 
x: 0     y: 3717     z: 7 
x: 0     y: 3721     z: 24 
x: 0     y: 3718     z: 25 
x: 0     y: 3714     z: 9 
x: 0     y: 3727     z: 10 
x: 0     y: 3721     z: 26 
x: 0     y: 3722     z: 26 
x: 0     y: 3727     z: 5 
x: 0     y: 3722     z: 26 
x: 0     y: 3729     z: 24 
x: 0     y: 3727     z: 12 
x: 0     y: 3727     z: 20 
x: 0     y: 3716     z: 24 
x: 0     y: 3705     z: 24 
x: 0     y: 3723     z: 15 
x: 0     y: 3689     z: 25 
x: 0     y: 3718     z: 24 
x: 0     y: 3721     z: 24 
x: 0     y: 3716     z: 10 
x: 0     y: 3736     z: 21 
x: 0     y: 3718     z: 26 
x: 0     y: 3715     z: 26 
x: 0     y: 3629     z: 22 
x: 1673     y: 1117     z: 2216 
x: 1652     y: 1121     z: 2243 
x: 0     y: 3696     z: 25 
x: 0     y: 3693     z: 24 
x: 2319     y: 1389     z: 1952 
x: 2311     y: 1387     z: 2212 
x: 2291     y: 1369     z: 2234 
x: 2281     y: 1363     z: 2268 
x: 0     y: 3599     z: 20 
x: 0     y: 3712     z: 25 
x: 1746     y: 2118     z: 1741 
x: 1713     y: 2123     z: 1898 
x: 1770     y: 2116     z: 1920 
x: 1749     y: 2123     z: 1895 
x: 0     y: 3775     z: 24 
x: 0     y: 3646     z: 24 
x: 0     y: 3695     z: 13 
x: 1687     y: 2163     z: 1854 
x: 1696     y: 2157     z: 1875 
x: 1694     y: 2157     z: 1896 
x: 1693     y: 2155     z: 1901 
x: 1693     y: 2157     z: 1920 
x: 0     y: 3698     z: 25 
x: 0     y: 3704     z: 24 
x: 0     y: 3711     z: 22 
x: 1841     y: 2179     z: 1664 
x: 1843     y: 2178     z: 1876 
x: 1841     y: 2176     z: 1888 
x: 1841     y: 2180     z: 1920 
x: 1839     y: 2179     z: 1908 
x: 1830     y: 2170     z: 1786 

Touch_calibrate.ino Output
// Use this calibration code in setup():
  uint16_t calData[5] = { 271, 3658, 237, 3581, 7 };
  tft.setTouch(calData);

All corner touch done.

Please see the videos on the following link.
https://1drv.ms/u/s!AlcBNDPDdJHVmBhNFjyUCJsw4zKR?e=om8Opj

Attached Keyboard project  is needed. 
[image]

copyCATandpasteDOG, 2022-09-16:
Hi Sudupatil,
check if there has been a tft rotation,in this case you have to erase the SPIFFS and
do a calibration again,as it is recommended:
// Set the rotation before we calibrate
  tft.setRotation(0);

Sudupatil, 2022-09-17:
Hello Bodmer,

Today I tried many times,  after multiple time making connection again and again; i fail to even get last result. so i changed the all jumper cables and breadboard pin location. and everything working properly. I don't need to change screen rotation. but calibration is required.

so i can conclude that if it's not working then try again rewiring. there is no problem with library (I never said it has).  In feature whoever facing this problem make sure all pin configuration and rewiring couple of times.

Please have look at the video of working.
https://1drv.ms/v/s!AlcBNDPDdJHVmBwQtqH29iOQ28ue?e=iBo9m0

Thank you so much for helping me out. Have good day ahead.
Sudarshan Patil

Bodmer, 2022-09-17:
Well done for being persistent and getting it working. I hope you have fun with the library now after all the frustration with the wire connections.

commit 2b576dc6 (2022-09-14): Fix #2018
closed 2022-09-17 by Bodmer