Bodmer/TFT_eSPI #2022: [TTGO T8 ESP32-S2] Screen remains blank

Shohreh, 2022-09-14:
Hello,

I'm a newbie at micro-controllers, si it could be something obvious to experienced developpers.

While [the board](http://www.lilygo.cn/prod_view.aspx?TypeId=50062&Id=1321) successfully sends text through the serial port*, the screen remains hopelessly blank when trying different scripts under c:\Users\blah\Documents\Arduino\libraries\TFT_eSPI\examples\.

Does TFT_eSPI supports that chip? Any idea what I could try?

Thank you.

*
```
void setup() 
{ 
	Serial.begin(9600); 
	Serial.println("In setup"); 
} 
  
void loop() 
{
	Serial.println("In loop"); 
	delay(500);
}
```

Bodmer, 2022-09-14:
Have you created a setup file for the display interface configuration?

You can run the Read_User_Setup sketch and see the used setttings.

Shohreh, 2022-09-15:
Here it is.

The screen is still dark, but I guess it's to be expected and the script only returns infos about the HW:

[code]
TFT_eSPI ver = 2.4.72
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 23
MISO    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 0
TFT_RST  = GPIO 2

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]

Shohreh, 2022-09-17:
As a total newbie, I edited User_Setup.h thusly, but the display remains dark:

```
#define ST7789_DRIVER
#define TFT_WIDTH   240
#define TFT_HEIGHT 320
#define TFT_MISO 23
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS     15
#define TFT_DC       0
#define TFT_RST   2
```
I compiled a random test script (TFT_Ellipse.ino).

Leaving... Hard resetting via RTS pin...: NOTHING

Hit RST button:; NOTHING
Unplugged, repplugged: NOTHING

Provided the settings above are correct, does it mean that display isn't supported?

Shohreh, 2022-09-19:
Per https://iotips.tech/using-tft_espi-library-on-esp32/, I forgot to comment the NodeMCU lines in User_Setup.h's Section 2… but TFT_Ellipse.ino = No change.

Next, per https://github.com/Xinyuan-LilyGO/LilyGo-T-Display-S2/blob/master/main/app_main.c, I modifed the settings… screen still dark :-/

//NOTHING
//#define TFT_MISO 19
//#define TFT_MOSI 23
//#define TFT_SCLK 18
//#define TFT_CS     15   // Chip select control pin
//#define TFT_DC       2   // Data Command control pin
//#define TFT_RST     4   // Reset pin (could connect to RST pin)
//#define TFT_RST   -1   // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

//NOTHING
//Per Read_User_Setup.ino
/*
#define TFT_MISO 23
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS     15   // Chip select control pin
#define TFT_DC       0   // Data Command control pin
#define TFT_RST   2   // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
*/

//NOTHING
//Per https://raw.githubusercontent.com/Xinyuan-LilyGO/LilyGo-T-Display-S2/master/main/app_main.c
#define TFT_MISO 4
#define TFT_MOSI 35
#define TFT_SCLK 36
#define TFT_CS     34
#define TFT_DC       37
#define TFT_RST   38

Shohreh, 2022-09-20:
Also tried this, still dark:

//Per https://github.com/Xinyuan-LilyGO/ESP32_S2/issues/1#issuecomment-1139405353
#define TFT_MISO 35 //Assume same as MOSI
#define TFT_MOSI 35
#define TFT_SCLK 36
#define TFT_CS   34
#define TFT_DC    37
#define TFT_RST   38

Bodmer, 2022-09-20:
I had a look at the documentation for the board. Firstly the display is advertised as a ST77789 type, maybe that is a typo as the ST7789 is more common. The library only supports the ST7789 type. Secondly the interface has no DC pin so it is not compatible with the interface type supported by the library.

Therefore I conclude that it will not work with this library.

closed 2022-09-20 by Bodmer

Shohreh, 2022-09-20:
Too bad :-/

I'll get another board, after checking that it's supported by TFT_eSPI.

Thanks much.

lyusupov, 2022-10-02:
https://github.com/Bodmer/TFT_eSPI/pull/1531

Bodmer, 2022-10-03:
I had a further look at the Lilygo documentation and it contains errors. The pin MISO is stated as being GPIO 37, but in fact GPIO 37 is the DC signal.

Also there appears to be no display type ST77789, so that is a typo also.

So in summary the display should work fine with  @lyusupov 's setup file.

Camsoh, 2022-10-08:
Can confirm that @lyusupov 's setup file works. 

Going to my TFT_eSPI library I (1) copied the new code into the Setup71 file and (2) adjusted the setup file in the User_Setup_Select file.

Thanks a bunch @Bodmer