Bodmer/TFT_eSPI #2028: [Question regarding FS ussed on espressif MCU]: For ESP32 the LITTLE FS is recommended now

hasenradball, 2022-09-16:
Hi,

is the actual release using the Little FS filesystem now.
From Gitter chat I got the info FS is deprecated now and LittleFS is recommended for ESP32.

Bodmer, 2022-09-17:
Yes, you need board package 2.0.0 or later.

closed 2022-09-17 by Bodmer

hasenradball, 2022-09-18:
Ok great, Thanks for answering.

I use the `TFT_eSPI` lib since years with the Espressif chips 8266 and 32.
Is there a big deal to switch to little fs in the `TFT_eSPI` lib?
The board package is use nearly the latest ones.

Bodmer, 2022-09-18:
LittleFS support is built in for smooth fonts, see examples. Sketches can also use LittleFS, see PNG example.

hasenradball, 2022-09-18:
But in `Extensions/Smooth_Font.h` SPIFFS is hard coded, right?

[image]

Thus switching to LittleFS there is a Warning:

[image]

I think if it is hardcoded in the FT_eSPI` lib then it has a difference between main file and lib.

hasenradball, 2022-09-23:
Can you please answer how to switch to little fs without warning?

Bodmer, 2022-09-24:
The warning can be ignored It is generated by the processor board package to encourage users to migrate to LittleFS. The library intentionally supports SPIFFS still so that legacy sketches continue to run so, at the moment, the warning gets reported.

hasenradball, 2022-09-25:
I thought I saw in the code that using SPIFFS is hardcoded.
Maybee I am wrong.
So I understand you that it is sufficient to use the LITTLEFS in the main sketch right?

Bodmer, 2022-09-25:
Yes, there are both LittleFS  and SPIFFS examples included in the library.

Don't forget to upload the required files to the processor using the IDE tools.

hasenradball, 2022-09-25:
I think you did not understand me right…
I found ich the src files a part were the SPIFFS is hardcoded. Not in the examples…

hasenradball, 2022-10-03:
Sorry again for the Stupid Question....
In the file `Smooth_font.h` there you will fin a hardcoded part with SPIFFS.
See line 45ff.
```
#ifdef FONT_FS_AVAILABLE
  fs::File fontFile;
  fs::FS   &fontFS  = SPIFFS;
  bool     spiffs   = true;
  bool     fs_font = false;    // For ESP32/8266 use smooth font file or FLASH (PROGMEM) array

#else
  bool     fontFile = true;
#endif
```
Do I have to change here when I use the LittleFS?

hasenradball, 2022-10-03:
What about the idea to use an #define in the User_Setup.h and do it like this:

```
#ifdef ESP8266
  #include "LittleFS.h"
  #define FILESYSTEM LittleFS
  void listDir(void);
#endif
```
And use the `FILESYSTEM` in all other files.
So you are able to select the filesystem by the user settings.

hasenradball, 2022-10-08:
Hi Bodmer,

can you please answer why in the `Smooth_font.h` is the Spiffs hardcoded and how is it working when I set the Filesystem to LittleFS?