Bodmer/TFT_eSPI #2029: Sprites on ESP32-WROOM-32E (DevkitC) + ST7789 SPI display not working?

alandsidel, 2022-09-17:
Got one of these displays today and started playing with it.  All is working pretty well with this library, thanks for all the effort that's gone into it!  I ran into the flickering everyone gets when updating the whole display and decided to try the sprites, but I haven't been able to get it working to save my life.  Simple working non-sprite example:

```
TFT_eSPI tft = TFT_eSPI();

tft.init();
tft.setRotation(1);
tft.fillScreen(TFT_BLACK);

tft.setTextColor(TFT_RED);
tft.setCursor(0, 0, 1);
tft.print("Hello World RED!");
```

This works as expected.  If I modify that to make a full screen sprite and send it in afterwards as follows, the output is unchanged when I expect it to show the sprite with the blue text and not the red text (thanks to not using transparency..)
```

TFT_eSPI tft = TFT_eSPI();
TFT_eSprite screen = TFT_eSprite(&tft);

tft.init();
tft.setRotation(1);
tft.fillScreen(TFT_BLACK);

tft.setTextColor(TFT_RED);
tft.setCursor(0, 0, 1);
tft.print("Hello World RED!");

screen.createSprite(tft.width(), tft.height());
screen.fillSprite(TFT_BLACK);
screen.fillScreen(TFT_BLACK);
screen.setTextColor(TFT_BLUE);
screen.setCursor(0, 16, 1);
screen.print("Hello World BLUE!");
screen.pushSprite(0, 0);
```

I've tried just about every combination I can think of to get this working, with and without the direct non-sprite writes, sprinkling `yield()` statements at various points, using one, both, or neither of `fillSprite` and `fillScreen`, moving the x/y coordinates of setCursor in case the origin of the text was off the screen, and so on.   I just can't get the sprite to appear.  I even gave up on text in case it was something to do with the inscrutable (for me, so far!) font system to no avail:

```
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite screen = TFT_eSprite(&tft);

tft.init();
tft.setRotation(1);
tft.fillScreen(TFT_BLACK);

screen.createSprite(tft.width(), tft.height());
screen.fillSprite(TFT_BLUE);
screen.pushSprite(0, 0);
```

I would expect that to create an entirely blue screen, but it seemingly does nothing.

Uncommented lines from my User_Setup.h:

```
#define USER_SETUP_INFO "User_Setup"
#define ST7789_DRIVER

#define TFT_SDA_READ
#define TFT_RGB_ORDER TFT_BGR
#define TFT_WIDTH  240
#define TFT_HEIGHT 320
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5
#define TFT_DC   27
#define TFT_RST  19

// Nothing below this line was modified by me, but I include it for completeness

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

```

Exact display used: https://smile.amazon.com/dp/B082GFTZQD?psc=1&ref=ppx_yo2ov_dt_b_product_details

TFT_eSPI version: 2.4.72 (from Arduino library manager)

Building in VS Code with the Arduino extension.

Bodmer, 2022-09-17:
Do other library sprite examples work?

alandsidel, 2022-09-17:
> Do other library sprite examples work?

I had not tried any before, so I just went through a few and yes, at least the few I tried do work.  I am trying to look at them for any important differences with my code.  The only thing sticking out now is the delay between the tft being setup with init, setRotation, and fillScreen -- and the actual usage of the sprite.  My code is doing all of this right away in `setup()` right now while the examples all seem to incorporate various and long delays between initialization and sprite usage.

I wouldn't think that would matter, but maybe I'm mistaken?

alandsidel, 2022-09-17:
> > Do other library sprite examples work?
> 
> the examples all seem to incorporate various and long delays between initialization and sprite usage.
> 
> I wouldn't think that would matter, but maybe I'm mistaken?

It's not this, removing the delays from the examples and they still work.  I'm digging in, commenting out all the other code in my own sketch to see if something is conflicting there.  It's using a second UART to talk to a GPS as well as using an I2C OLED display on different pins.  All of this works as is, along with the TFT_eSPI when *not* using sprites.

alandsidel, 2022-09-17:
I've tracked the problem down to something to do with the dimensions of the sprite.  The code does not work when I use `screen.createSprite(tft.width(), tft.height());` but it works fine when I set a smaller size, like dividing both dimensions by 2.  I've also tried it with a size slightly smaller than the screen, for example with a 300x200  sprite, and that works as well.  It does *not* work with `screen.createSprite(tft.width() - 1, tft.height() - 1);` which I thought might be the case.

I am trying to determine just what size causes it to start failing.

Bodmer, 2022-09-17:
There are notes on sizes here:
https://github.com/Bodmer/TFT_eSPI#sprites

Later versions of the board package, 2.0.0 or later, allocate more RAM to the sketch so full screen 240x320 sprites may work.

If createSprite returns a null pointer then sufficient RAM for the sprite is not available.

alandsidel, 2022-09-17:
Thanks @Bodmer , I was beginning to suspect low memory was the issue.  I tried enabling psram only to discover that these particular boards do not have any.  I'm unsure what the board revision is.  The only version marking I can find is on the bottom which states `ESP32_DevKitc_V4`.

The chip itself is an ESP32-WROOM-32E which according to the datasheet has 520K of SRAM.  It seems like this should be enough to handle my simple sketch plus the size of the buffer, but maybe I'm overlooking something.

alandsidel, 2022-09-17:
Indeed there does seem to be plenty of free memory.  I changed the code to print the result of `ESP.getFreeHeap()` before and after the call to `createSprite` and to check the result of that function as you mentioned.

```
  Serial.print("Free Heap: ");
  Serial.println(ESP.getFreeHeap());

  if (NULL != screen.createSprite(320, 240)) {
    Serial.println("Sprite created");
    Serial.print("Free Heap: ");
    Serial.println(ESP.getFreeHeap());

    screen.fillSprite(TFT_GREEN);
    screen.setTextColor(TFT_RED);
    screen.setCursor(12, 5);
    screen.print("Original ADAfruit font!");
    screen.pushSprite(0, 0);
  } else {
    Serial.println("Sprite creation failed");
  }

```

With the 300x200 size:
```
Free Heap: 375336
Sprite created
Free Heap: 255316
```

Bumping it to 320x240, it does indicate creation failed.

alandsidel, 2022-09-17:
Well think I found it.  I suspect there's nothing I can do about it besides bumping the color depth down, but if you have any thoughts please feel free to share.  `ESP.getMaxAllocHeap()` is returning a value of 125920, which is about 18K short of what I need for a 320x240x16 frame buffer.  Memory fragmentation from before my sketch kicks off I suppose?

Thanks for the assistance!

closed 2022-09-17 by alandsidel

Bodmer, 2022-09-17:
OK, the only other option is to use a ESP32 board with PSRAM fitted or an ESP32 S3 board.

alandsidel, 2022-09-17:
> OK, the only other option is to use a ESP32 board with PSRAM fitted or an ESP32 S3 board.

Thanks for your help, sorry for wasting your time with all this. :)

alandsidel, 2022-09-17:
Hey @Bodmer one more question.  I can't find an "elegant" way to set the color depth while looking through the source.  `createSprite()` doesn't take a color depth, and `changeColorDepth()` uses the preexisting dimensions.  The 16bit color depth doesn't seem to be alterable.  Is there not a way to initialize it with dimensions and depth in one setting?

It looks like adding such a feature would be pretty simple and I'm game to try it and send a PR but I don't want to bother if there's already a supported way that I've overlooked.