Bodmer/TFT_eSPI #2031: Can't get it working on ESP8266 D1 Mini

MintKind, 2022-09-17:
Hello everyone, im quite new to the whole space and need some help getting the library working on my ESP8266. The code im using was runnig fine on a ESP32 and did properly output to the display. Here is my current setup:

IDE: Arduino IDE
Board: ESP8266 D1 Mini ([Pinout](https://cdn.shopify.com/s/files/1/1509/1638/files/D1_Mini_Pinout_Diagram.pdf?5603303439559815202))
Display: Waveshare 1,28 inch round display (GC9A01) ([Wiki](https://www.waveshare.com/wiki/1.28inch_LCD_Module#Arduino))

**Connections:**
**_**Display -> Board**_**
VCC -> 5v
GND -> GND
DIN -> D7
CLK -> D5
CS -> D1 (not working on D8 as esteblished [here ](https://github.com/Bodmer/TFT_eSPI/discussions/1490#discussioncomment-2093509)
DC -> D3
RST -> D4
BL -> 3.3v

Here is my [User_Setup](https://pastebin.com/S8HTBC3M)

Everything compiles and transfers fine, but no code I throw at it workes. Just backlight, no image. Does anyone have an idea what could cause this?

Bodmer, 2022-09-17:
Try using GPIO numbers rather than Dx.

MintKind, 2022-09-17:
In User_Setup I changed it to this:
```
#define TFT_CS   5  // Chip select control pin D8
#define TFT_DC   0  // Data Command control pin
#define TFT_RST  2  // Reset pin (could connect to NodeMCU RST, see next line)
```
This should be the correct GPIO numbers acording to the pinout. But sadly still just backlight

MintKind, 2022-09-17:
Dont know what did it, but after a lot of flashes and trying different pin configurations this worked for me:

**Connections:**
_Display -> Board_
VCC -> 3.3v
GND -> GND
DIN -> D7
CLK -> D5
CS -> D0 (not working on D8 as esteblished here
DC -> D3
RST -> D4
BL -> not connected

**Config:**
#define GC9A01_DRIVER
#define TFT_HEIGHT 240
#define TFT_CS PIN_D8
#define TFT_DC PIN_D3
#define TFT_RST PIN_D4

closed 2022-09-17 by MintKind

Bodmer, 2022-09-17:
Glad you have it working.

hasenradball, 2022-09-18:
Hi here you will see my Configuration which I use with the Wemos D1 mini:

```
// See SetupX_Template.h for all options available

#define ILI9341_DRIVER

// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP8266 SETUP ######

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation

// Master Out Slave In
#define TFT_MOSI 13
// Master In Slave Out
#define TFT_MISO 12
// SPI Clock
#define TFT_SCLK 14
// Chip Select or Slave Select
#define TFT_CS 16
// Data Command Control
#define TFT_DC 15
// Reset
//#define TFT_RST -1

```