Bodmer/TFT_eSPI #2047: User_Setup Pin assignments not being applied

Bwanna, 2022-09-27:
(Trying to follow the template. Please let me know if I missed anything.)

1. ISSUE DESCRIPTION: 
- Pin assignments in User_Setup.h are not being compiled in sketch when tested with "Read_User_Setup" (DC and RST are not correct)
- Pin assignments placed in sketch are not being compiled either (same output observed)
3. IDE Arduino v2.0.0
4. TFT_eSPI library version 2.4.72
5. Board package version (don't know how to find this info - get from espressif)
6. ESP32 DevKitC
7. TFT driver ST7789
8. Interface type SPI

**User_Setup.h**
```
// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
```
**Read_User_Setup.h OUTPUT**
```
TFT_eSPI ver = 2.4.72
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 240
Display height = 320

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 0
TFT_RST  = GPIO 2

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
```

Bodmer, 2022-09-29:
Check DC and RST pins are not being defined elsewhere and that the correct setup file is being included.

Bwanna, 2022-09-29:
**Here's what I've done, still not working:**

- Did a fresh reinstall of Arduino IDE v2.0.0.
- Installed board URL to espressif
- installed fresh library for TFT_eSPI v2.4.76,
- Changed ONLY the necessary settings in both User_Setup_Select.h and User_Setup.h
- used "Setup18_ST7789" (have tried other Setups with no luck)
- board is ESP32-WROOM-32 (devkit)

Have tested with a few different ESP dev modules - no luck.   
Again - hardware works fine using any other ST7789 library with the same configurations.  Need to get TFT_eSPI working so it will interface with GUISlice software.

ANY ideas would be appreciated.

Bwanna, 2022-09-29:
**FIXED**

Had to insert into Setup18_ST7789.h the pin defines.   
Don't understand why the same pin defines wouldn't work when placed in the Sketch or the User_Setup.h file?

closed 2022-09-30 by Bodmer