Bodmer/TFT_eSPI #2053: ERROR: "class TFT_eSPI' has no member named 'getTouchRaw'"

Bwanna, 2022-09-29:
**1. ERROR  "Compilation error: 'class TFT_eSPI' has no member named 'getTouchRaw'"**
2. IDE Arduino v2.0.0
3. TFT_eSPI library version 2.4.72
4. Espressif ESP url
5. ESP-WROM-32 (DevkitC)
6. TFT ST7789  SPI 4-wire
7. STMPE610 I2C

Full Error output Msg: Test_Touch_Controller.ino

```
D:\Arduinofiles\libraries\TFT_eSPI\examples\Test and diagnostics\Test_Touch_Controller\Test_Touch_Controller.ino: In function 'void loop()':
D:\Arduinofiles\libraries\TFT_eSPI\examples\Test and diagnostics\Test_Touch_Controller\Test_Touch_Controller.ino:36:7: error: 'class TFT_eSPI' has no member named 'getTouchRaw'
   tft.getTouchRaw(&x, &y);
       ^~~~~~~~~~~
D:\Arduinofiles\libraries\TFT_eSPI\examples\Test and diagnostics\Test_Touch_Controller\Test_Touch_Controller.ino:42:33: error: 'class TFT_eSPI' has no member named 'getTouchRawZ'
   Serial.printf("z: %i \n", tft.getTouchRawZ());
                                 ^~~~~~~~~~~~

exit status 1

Compilation error: 'class TFT_eSPI' has no member named 'getTouchRaw'
```

**Here are the setup files:**
[TFT_eSPI USer_Setup Files.zip](https://github.com/Bodmer/TFT_eSPI/files/9679364/TFT_eSPI.USer_Setup.Files.zip)

I'm still new at this, but it appears the "Touch.h" is not being referenced somehow?  I can see both of these functions are in that file.

Thanks!

Bodmer, 2022-09-30:
Correct, the library doesnot support I2C for the touch controller. There are other touch libraries that can be used.

closed 2022-09-30 by Bodmer

Bwanna, 2022-09-30:
How should the other touch library (Adafruit_STMPE610.h) be used within TFT_eSPI so that everything works?  I've tried including it in the sketch, but get the same error.

Bodmer, 2022-09-30:
Have you commented out the #define TOUCH_CS in the TFT_eSPI setup file?