Bodmer/TFT_eSPI #2055: Replace hardcoded SPIFFS by Parameter

hasenradball, 2022-10-03:
Sorry again for the Stupid Question....
In the file `Smooth_font.h` there you will fin a hardcoded part with SPIFFS.
See line 45ff.
```
#ifdef FONT_FS_AVAILABLE
  fs::File fontFile;
  fs::FS   &fontFS  = SPIFFS;
  bool     spiffs   = true;
  bool     fs_font = false;    // For ESP32/8266 use smooth font file or FLASH (PROGMEM) array

#else
  bool     fontFile = true;
#endif
```
Do I have to change here when I use the LittleFS?

And additionally in line 106.

_Originally posted by @hasenradball in https://github.com/Bodmer/TFT_eSPI/issues/2028#issuecomment-1265125016_

hasenradball, 2022-10-03:
What about the idea to use an #define in the User_Setup.h and do it like this:

#ifdef ESP8266
  #include "LittleFS.h"
  #define FILESYSTEM LittleFS
  void listDir(void);
#endif

And use the FILESYSTEM in all other files.
So you are able to select the filesystem by the user settings.
Just replaced the hardcoded SPIFFS by FILESYSTEM and you are finished!

hasenradball, 2022-10-03:
I use such an preprocessor define in my own spiffs function to select automatically the filesystem depending by the controller.

copyCATandpasteDOG, 2022-10-06:
Hi,it's a handy way to use the filesystems :) ,
but i have trouble with the LittleFS on pico boards,i have errors like :FS.h not found,and its seems the library
updated to 2.05 and the name is "lfs",no more LittleFS,but can you clear it to me,or someone else?

hasenradball, 2022-10-06:
@copyCATandpasteDOG Hi can you paste you code part which makes trouble?

copyCATandpasteDOG, 2022-10-06:
Thanks for your help :)
Here it is,my code trouble comes from a littlefs definition to store the calibration data for a PI PICO RP2040 & a TFT touch,
and the error shows: FS.h, no such file...Here is the code:

//------------------------------------------------------------------------

/**
  Title: Raspberry Pi Pico Touch Macro Keyboard Demo
  Author: Dustin Watts
  Date: 17-05-2021
  A simple 4 by 3 matrix of buttons. To show you how to create a custom marco keyboard. Each button can have it's
  own function. The button presses are handled in buttonpress().
  Using Earle Philhower's 'arduino-pico': https://github.com/earlephilhower/arduino-pico . This core comes with LittleFS and a Keyboard library.
  Make sure to flash your Pico with at least 64KB for the filesystem!
  This example is based on the examples that come with the TFT_eSPI library. Which includes the Adafruit GFX button handling class.
  These libraries need to be installed:
  - TFT_eSPI (available through the Library Manager)
  ! Don't forget to configure User_Setup.h in the TFT_eSPI library!
  
    TFT_MISO  0
    TFT_MOSI  3
    TFT_SCLK  2
    TFT_CS   20
    TFT_DC   22
    TFT_RST  21
    TOUCH_CS 14
*/

#include <FS.h>
#include <LittleFS.h>
#include <Keyboard.h>
#include <SPI.h>
#include <TFT_eSPI.h>

// Name of the Touch calibration file
#define CALIBRATION_FILE "/TouchCalData2"

// Define our filesystem
#define FILESYSTEM LittleFS

// Set to tru if you want the calibration to run on each boot
#define REPEAT_CAL false

// Keypad start position, key sizes and spacing
#define KEY_W 95 // Key width
#define KEY_H 95 // Key height
#define KEY_SPACING_X 20 // X gap
#define KEY_SPACING_Y 10 // Y gap

#define KEY_X (KEY_W/2) + KEY_SPACING_X // X-axis centre of the first key
#define KEY_Y (KEY_H/2) + KEY_SPACING_Y // Y-axis centre of the first key

#define KEY_TEXTSIZE 3   // Font size multiplier

// Choose the font you are using
#define LABEL1_FONT &FreeSansOblique12pt7b // Key label font

// Adding a delay between keypressing to give our OS time to respond
uint8_t keydelay = 100;

// Create the screen object
TFT_eSPI tft = TFT_eSPI();

// Creating the labels for the buttons
// <name>[<number-of-lables>][<number-of-chars-per-label]
// The number of chars per label should include the termination \0.
char keyLabel[12][3] = {"1", "2", "3", "4", "5", "6", "7", "8", "9", "10", "11", "12"};

// Setting the colour for each button
// You can use the colours defined in TFT_eSPI.h
[... 226 lines cut ...]
    tft.println();

    if (REPEAT_CAL) {
      tft.setTextColor(TFT_RED, TFT_BLACK);
      tft.println("Set REPEAT_CAL to false to stop this running again!");
    }

    tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);

    tft.setTextColor(TFT_GREEN, TFT_BLACK);
    tft.println("Calibration complete!");

    // store data
    File f = FILESYSTEM.open(CALIBRATION_FILE, "w");
    if (f) {
      f.write((const unsigned char *)calData, 14);
      f.close();
    }
  }
}

//---------------------------------------------------------------------------------------

If you have a PICO around maybe you can try it and see if you have some compiling error
https://github.com/earlephilhower/arduino-pico

copyCATandpasteDOG, 2022-10-06:
That's sad,anyway, thanks for help :)

hasenradball, 2022-10-06:
have you tried to comment the line:

#include <FS.h>

?

Bodmer, 2022-10-08:
I do not understand the issue. If you are getting errors with the examples then please explain fully the erreos and your setup.

If you want to use SPIFFS for the font files then base the code on [these examples](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Smooth%20Fonts/SPIFFS) which are in the SPIFFS examples submenu.

If you want to use LittleFS for the font files then base the code on [these examples](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Smooth%20Fonts/LittleFS) which are in the LittleFS examples submenu.

As far as I can see there is no need to add further complexity.

hasenradball, 2022-10-09:
Hi I don't want to add not needed complexity.

I just want to understand why in the `Smooth_Font.h` file (See line 45ff.) the Spiffs is hardcoded there.
Does it not conflict when you set the filesystem in `main.ino` to LittleFS and in the `Smooth_Font.h` it is hardcoded to Spiffs?

Bodmer, 2022-10-09:
OK, I can see you did not understand my previous answers.

1. SPIFFS is NOTt hard coded because the FS variable fontFS  can be modified at runtime.
2. SPIFFS is set as the default for the variable fontFS, just in the same way you might declare a numeric variable as 0 initially so it is not undefined.
3. By setting fontFS by default to equal SPIFFS allows legacy scketches (those implemented by users before LittleFS was added to a board pacjkage) and the use of historic board packages that did not support LittleFS. Some users do not upgrade to the latest board package due to bugs or configuration managment reasons. Clearly, setting fontFS by default to LittleFS would break those sketches.
4. 2. A function is provided to set the FS, viz tft.loadFont(AA_FONT_SMALL, LittleFS);, this is why SPIFFS and LittleFS examples are slightly different. Equally you could use tft.loadFont(AA_FONT_SMALL, SPIFFS);

Please only raise issues for real problems with the library. Raise a pull request for inprovement suggestions and the discussions for general questions.

closed 2022-10-09 by Bodmer