Bodmer/TFT_eSPI #2065: compile error ESP32-S2 and ST7789

mmstac, 2022-10-07:
1. Getting a compile error when selecting any ESP32-S2 board

TFT_eSPI_ESP32.c:74:31: error: 'VSPIQ_IN_IDX' was not declared in this scope
   pinMatrixInAttach(TFT_MOSI, VSPIQ_IN_IDX, false);
............................................................^~~~~~~~~~~~
note: suggested alternative: 'FSPID_OUT_IDX'

3. IDE Arduino 1.8.19
4. TFT_eSPI v2.4.72
5. Board 2.0.5
ESP32-S2 ST7789v2 SPI
[tft_setup.h.txt](https://github.com/Bodmer/TFT_eSPI/files/9737552/tft_setup.h.txt)
[error.txt](https://github.com/Bodmer/TFT_eSPI/files/9737555/error.txt)

Bodmer, 2022-10-08:
This does not make any sense, as long as this line:
```
//#define TFT_SDA_READ
```
 is commented out then the error will not be reported as that code will not be compiled. [See here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.c#L62-L91).

mmstac, 2022-10-09:
Thanks for you reply (and thank you for sharing your great library). I tested again and it is working. Seems I somehow lost my platforms.local.txt and I did not realize it was not using the sketch's tft_setup.

closed 2022-10-09 by mmstac