Bodmer/TFT_eSPI #2072: Support display on Lilygo T-QT (ESP32S3)

eni23, 2022-10-11:
In order to get the display on a [Lilygo T-QT](https://www.lilygo.cc/products/t-qt-v1-1) working with TFT_eSPI, there is a change in `TFT_Drivers/GC9A01_Rotation.h` needed, otherwise there is a weird border and the display does not really displays the whole area.

The changed file can be found [here](https://github.com/Xinyuan-LilyGO/T-QT/blob/main/lib/TFT_eSPI/TFT_Drivers/GC9A01_Rotation.h).

It would be nice to integrate this change in TFT_eSPI upstream somehow so there won't be any need for maintaining a separate repo. Since just changing the file would probably break something else, i'm opening this issue here to ask how it could be done.

Thanks :)

Bodmer, 2022-10-12:
The module seems to have a new display controller type GC9107. This appears to have the same commands and register init values as the GC9A01 but has different offsets since not all the pixel capability is used.  I think this ideally needs another driver file set to avoid confusion. e.g. #define GC9107_DRIVER. This is simple to implement and I may be able find time to update the library the near future. In the meantime I see you have a solution.

I assume the display works correctly in all possible rotations 0-3 with the proposed setup file? Correct?

eni23, 2022-10-13:
> I assume the display works correctly in all possible rotations 0-3 with the proposed setup file? Correct?

Yes, all rotations work fine.

> I think this ideally needs another driver file set to avoid confusion. e.g. #define GC9107_DRIVER. This is simple to implement and I may be able find time to update the library the near future

If it is just adding the files `TFT_Drivers/GC9107_Init.h`,  `TFT_Drivers/GC9107_Rotation.h`,  `TFT_Drivers/GC9A01_Defines.h` and a user-setup (which number should i choose?) and adjusting them accordingly, i can prepare a MR as well.

One thing i was wondering: Lilygo has a different display example whrere they do not need your library but a "custom driver". There the init commands look quite different than the ones in `GC9A01`: https://github.com/Xinyuan-LilyGO/T-QT/blob/main/example/lcd/lcd_driver.cpp#L109

But the ones from `GC9A01` seem to work perfectly fine so i assume just keep going with those ones?

Bodmer, 2022-10-13:
It is best to use the specific setup recommended b the vendor to get the best out of the display in terms of colour fidelity and Gamma balance. The Lilygo code is useful as it can be adapted.

The driver files for init and rotation need to be called up in TFT_eSPI.cpp as well, the location is place marked circa line 700 and line 800. Also in User_Setup_Select.h circa line 200.