Bodmer/TFT_eSPI #2073: TFT has colors (R and B swapped).  How do i create the correct hex

geetee24, 2022-10-12:
how do i create the correct 4 char hex for each TFT_COLOR RGB?

Bodmer, 2022-10-12:
I assume the displayed colours are wrong? Assuming the answer is yes:

There is a control register that sets the TFT to either RGB or BRG colour order, this permits manufacturers to wire the pixel R and B colours as they wish. Since the colour order varies between suppliers there is an option (only for some display types currently) built into the library to specify the order in the setup file, example here:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L72-L78

Another interpretation of the question:
There is also a function:
```
uint16_t color565(uint8_t red, uint8_t green, uint8_t blue);
```
to convert 8 bit RGB values to a 16 bit colour if that is what you are after.

closed 2022-10-12 by Bodmer

geetee24, 2022-10-12:
Cool.  any sample code on how to write to that control reg?

Bodmer, 2022-10-12:
I don't recommend writing direct to that register as it has other bits that must be set correctly and these are dependant on the TFT controller used and rotation set.

Example code to write the the MADCTL in an ILI9341 display register here:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9341_Rotation.h

MADCTL register bit definitions for different controllers, consult the datasheet for your controller.

For ST7735, ST7789 and ILI9341 it is possible to defined the colour order in the setup file as noted above.

Bodmer, 2022-10-12:
Here is the function:
void     writeRegister(uint16_t c, uint8_t d); // Write data to 16 bit command register