Bodmer/TFT_eSPI #2078: New WT32-SC01-PLUS display not supported due to parallel pins out of range

fritsjan, 2022-10-13:
Unfortunately the pins used in the latest sucessor of the WT32-SC01 display, the WT32-SC01-PLUS which uses a parallel display, some of the pins used are out of the range of 0-31...
It took me a day to figure this out why it just did not work, this is the setup of the pins:

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
#define TFT_PARALLEL_8_BIT

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   -1  // Chip select control pin (library pulls permanently low)
#define TFT_DC   0  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  4  // Reset pin, toggles on startup

#define TFT_WR    47  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    48  ??  // Read strobe control pin

#define TFT_D0   9  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   46  // so a single register write sets/clears all bits.
#define TFT_D2   3  // Pins can be randomly assigned, this does not affect
#define TFT_D3   8  // TFT screen update performance.
#define TFT_D4   18
#define TFT_D5   17
#define TFT_D6   16
#define TFT_D7   15

As you can see TFT_D1 and TFT_WR, TFT_RD are all above 31....
Therefore this display cannot be used with TFT_eSPI, which is a shame because the display is very very promising...

Is there a workaround?

Bodmer, 2022-10-13:
The library would need adaption, that combination of GPIO for the data bus across multiple control registers is very inconvenient.  It is unlikely I will be making changes to accommodate the board. Many others are supported and I do not have time to support every product. Sorry!

The GPIO allocation constraints are in the comments, but when I finally get to writing a User Manua that will make it clearer.
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup13_ILI9481_Parallel.h#L11-L25

closed 2022-10-13 by Bodmer

elgerg, 2022-12-29:
Oh this is a shame. I just got one of these and was really hoping I could use the TFT_eSPI library :(

kraa965, 2023-02-12:
> Oh this is a shame. I just got one of these and was really hoping I could use the TFT_eSPI library :(

hi, were you able to figure out how to run WT32-SC01-PLUS on this library?

fritsjan, 2023-02-12:
No, due to using certain pins on this display it is impossible.
But have a look here to get the display working: https://github.com/fritsjan/WT32-SC01-PLUS-PLATFORMIO

kraa965, 2023-02-12:
> No, due to using certain pins on this display it is impossible. But have a look here to get the display working: https://github.com/fritsjan/WT32-SC01-PLUS-PLATFORMIO

Thank you, I saw it, but with my project why it doesn't start, I don't understand what the error is

fritsjan, 2023-02-12:
You mean this library?
I also thought I did everything correct. And it gives no error. Only just don't work. It is because this lib can only use pins below a certain number. And the display unfortunately uses two pins above this number and therefore it does not work.

kraa965, 2023-02-12:
> You mean this library? I also thought I did everything correct. And it gives no error. Only just don't work. It is because this lib can only use pins below a certain number. And the display unfortunately uses two pins above this number and therefore it does not work.

No, I'm talking about your link https://github.com/fritsjan/WT32-SC01-PLUS-PLATFORMIO I tried the LovyanGFX library, the code compilation goes well, but the board does not show signs of life after a reboot, I do not understand what this is related to