Bodmer/TFT_eSPI #2079: Need super large font

geetee24, 2022-10-13:
Any advice on how to find a super large font?  Say 100 pixel tall?  I have a 220/330 tft that I want to show the temperature  so show it as 3 digits and viewable from say 15 feet away?

Or any URLs that tell me how to create my own?  Or maybe I have BMPs of 0-9 digits and then code to show say for 98 degrees, I show the BMP for 9 and 8 side by side?

thanks :)

Bodmer, 2022-10-13:
The largest font include is 72 pixels with digits only.

Options are:
1. Create a new custon Adafruit_GFX format font, search Google for online generation tools
2. Create a "Smooth font" with Processing IDE tools provided
3. Use the fonts available with this extension library : https://github.com/Bodmer/U8g2_for_TFT_eSPI
4. Use the fonts available with this extension library: https://github.com/Bodmer/TFT_eSPI_ext

Try to only include the characters you need otherwise the font file will be huge. Option 2 is the one I would try, Smooth fonts are generated using a Processing IDE utility in the library Tools folder. You can select the character subset included, font and size. See examples for their use.

closed 2022-10-13 by Bodmer

geetee24, 2022-10-13:
where do i get Processing IDE tools provided

geetee24, 2022-10-13:
i looked at 3 and 4.  but which font is the huge one?

geetee24, 2022-10-13:
where is the "The largest font include is 72 pixels with digits only."