Bodmer/TFT_eSPI #2080: SSD1963 - ESP32 - 8 bit Parallel - error: 'dir_mask' was not declared in this scope

iw7dmh, 2022-10-13:
Hi,

I configured the TFT_eSPI library for SSD1963 - 8 bit parallel on ESP32 but I am getting the following compile errors:

lib/TFT_eSPI-2.4.78/TFT_eSPI.cpp:994:10: error: 'dir_mask' was not declared in this scope
lib/TFT_eSPI-2.4.78/TFT_eSPI.cpp:1079:10: error: 'dir_mask' was not declared in this scope
lib/TFT_eSPI-2.4.78/TFT_eSPI.cpp:1205:10: error: 'dir_mask' was not declared in this scope

I found the variable 'dir_mask' isn't define anywhere. I changed che variable name in 'xdir_mask' as defined in TFT_eSPI.h file. Now the code compiles fine but I didn't tested it.

Can you confirm the problema and the correction?

Bodmer, 2022-10-13:
Hi, thanks for reporting this. I forgot to update that file after a global search and replace. I have fixed this now.

closed 2022-10-13 by Bodmer

Rohidfkra, 2022-10-20:
> Hi, thanks for reporting this. I forgot to update that file after a global search and replace. I have fixed this now.

Hello sir, I have tried to configure TFT_eSPI library for parallel ILI9486-8 bit on ESP32 but I getting some problems. I found the variable 'dir_mask' isn't defined anywhere.

Do you have update that file after a global search and replace? please help me 

Thank you very much

iw7dmh, 2022-10-20:
Hi friend, 
you can download again the TFT_eSPI zip archive. The working code is already in there. 
I was able to compile the code the day after I opened the issue. 
Best regards 
--
Inviato da Libero Mail per Android giovedì,

commit f106fb03 (2022-10-13): Fix #2080

Forgot to copy over updated file!