Bodmer/TFT_eSPI #2103: Sprites: 16-bit color and gradients aren't working on Seeeduino Xiao

cjstangman, 2022-10-21:
I am attempting to use sprites on a Seeeduino Xiao interfacing to a ST7735 driver in order to avoid flickering. I have full functionality if I write directly to the screen (Full colors and gradients), but if I try to set the sprite color depth to 16, I get a blank sprite, and even with a depth of 4, the gradients are empty(or black, not sure).

This could be a memory limitation, but that seems unlikely to me as the Arduino IDE is telling me that I am only using 38% of storage.

Bodmer, 2022-10-21:
The Seeed Studio XIAO is available with different processors, so if you are using the SAMD21 note it has only 32kbytes of RAM.

An alternative guess is that there may be a problem with the way the sketch is written.

cjstangman, 2022-10-21:
If you have seen the gradients work on sprites in other processors, it is probably not the library. I am using the SAMD21 XIAO, and given that it works somewhat with 4 bit colors, but not with 16 bit ones, a memory issue is most likely. I didn't know about the other XIAO processor options though, so thanks for that!

cjstangman, 2022-10-27:
Hey Bodmer,

So, I managed to get a smaller sprite to work with 16-bit color, but the gradient rectangles still don't do anything. Regular rectangles work, and the gradients work when writing directly to the screen, but they just show up black on sprites. 

Any ideas? The only thing that is different is the alpha blending, no?

Anyway, I can probably survive with solid colors, but the gradient definitely looks better and I'd love to use it.

Thanks

Bodmer, 2022-10-27:
Post a simple sketch that demonstrates the problem.

cjstangman, 2022-10-27:
I can, but I was just looking at the code, and fillRect is declared as virtual in TFT_eSPI.h and overridden in Sprite.h/cpp, but fillRectVGradient isn't. I assume that that is the issue.

Bodmer, 2022-10-29:
The gradient in a sprite works fine for me, so post a simple sketch that demonstrates the problem and I will take a look.

fillRectVGradient does not need to be virtual, it is inherited by the sprite class and the graphics functions used inside fillRectVGradient are virtual.

cjstangman, 2022-10-30:
Ok, so I wrote this sketch, but it is not showing up as black, but rather showing up in weird places.

Run it as written and on my display the sprite at 30, 30 changes to the gradient one, and the new sprite at 80, 30 is the green one.

If you comment out Section 1, the sprite push to 80, 30 displays as a very thin stripe across half of the top of the screen and there is a white stripe across the top of the other half.

```#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>

#define LIGHT_GREEN         0x87cb
#define DARK_GREEN          0x1681
#define PURPLE              0x8170

#define BKLT_PIN            7

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h
TFT_eSprite spr = TFT_eSprite(&tft);

void setup(void) {
  pinMode(BKLT_PIN, OUTPUT);
  
  digitalWrite(BKLT_PIN, LOW);
    
  //Init Screen
  tft.init();
  tft.setRotation(3);
[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/9894957/User_Setup.zip)

  spr.setColorDepth(16);

  // Create a sprite of defined size
  spr.createSprite(20, 20); 

  tft.setSwapBytes(true);
  // push the image - only need to do this once.
  tft.fillScreen(TFT_BLACK);
  
  digitalWrite(BKLT_PIN, HIGH);

  //Section 1
  spr.fillRect(0, 0, 20, 20, LIGHT_GREEN);
  spr.pushSprite(30, 30);
  //End
  
  delay(1000);

  //Section 2
  spr.fillRectVGradient(0, 0, 20, 20, LIGHT_GREEN, PURPLE);
  spr.pushSprite(80, 30);
  //End
}

void loop() 
{
}
```

cjstangman, 2022-10-30:
[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/9894958/User_Setup.zip)

Bodmer, 2022-11-03:
Now fixed in master, thanks for reporting this.

commit d3a715da (2022-11-03): Fix #2103 and update for latest ESP8266 board package

Fix H and V gradient use in sprite
New ESP8266 board package uses RDUINO_ARCH_ESP8266
old package defined ESP8266
closed 2022-11-03 by Bodmer