Bodmer/TFT_eSPI #2106: fillRect() doesn't work when given negative width or height value (it works when using adafruit lib)

awocrf, 2022-10-22:
fillRect() doesn't show anything on the display when given negative width or height value. It is working when I'm using the Adafruit library.

PlatformIO
v2.4.78
RP2040 uC
ILI9341
SPI

Bodmer, 2022-10-23:
Yes, negative widths, heights and radii are rejected in this library for graphics, images, sprites and viewports. I won't be changing this so extra code will be needed in your sketch to cover your use case.

closed 2022-10-23 by Bodmer

Bodmer, 2022-10-23:
I think the Adafruit library is inconsistent in the behaviour with negative widths and/or heights, sometimes it draws with incorrect size, in other cases it draws nothing. This suggests the Adafruit_GFX library behaviour with negative values is by accident and not by design.

I believe the TFT_eSPI library behaves consistently and is therefore correct.