Bodmer/TFT_eSPI #2109: SPI pin definitions are missing for RP2040 when using ST7735-based 128x128 LCD with SPI1

vishnumaiea, 2022-10-24:
**What I am using:**

**Board:** [Wiznet WizFi360-EVB-Pico](https://docs.wiznet.io/Product/Open-Source-Hardware/wizfi360-evb-pico)
**Processor:** RP2040
**LCD:** [1.44", 128x128, ST7735](https://www.tindie.com/products/icstation/5v-tft-144-inch-color-lcd-for-arduino8542/)
**Interface:** ```SPI1```
**Platform:** Arduino (Earle's RP2040 Arduino package)
**TFTeSPI Library version:** 2.4.72

**What I am doing:**

I am trying to use an ST7735-based 128x128 1.44" LCD with RP2040 by wiring it to the hardware `SPI1` port. The board is a **WizFi360-EVB-Pico** from WizNet. The board is identical to the official Raspberry Pi Pico board with the difference being the addition of the WizFi360 module connected to `Serial2` port. So nothing on the way of TFTeSPI or LCD.

Below is the TFTeSPI configuration I am using. Only active configuration lines are included. Everything else is commented out.

```
#define ST7735_DRIVER      // Define additional parameters below for this display

#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

#define TFT_WIDTH  128
#define TFT_HEIGHT 128

#define ST7735_GREENTAB3

#define TFT_CS     7   // Chip select control pin D8
#define TFT_DC     11  // Data Command control pin (RS pin)
#define TFT_RST    10  // Reset pin (could connect to NodeMCU RST, see next line)
// #define TFT_MOSI   15
// #define TFT_MISO   12
// #define TFT_SCK    14
// #define TFT_SCLK   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// For the RP2040 processor define the SPI port channel used (default 0 if undefined)
#define TFT_SPI_PORT 1 // Set to 0 if SPI0 pins are used, or 1 if spi1 pins used

#define SPI_FREQUENCY  27000000
```

**The issue:**

So the issue is when I compile I get a **missing definition error** for two pins. Below is a section of the build log that lists the errors.

```
In file included from D:\Code\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:
D:\Code\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:882:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
882 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
|        ^~~~~~~
In file included from D:\Code\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:32:
D:\Code\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_RP2040.c:18:66: error: 'TFT_SCLK' was not declared in this scope; did you mean 'TFT_CS'?
18 |     SPIClassRP2040 spi = SPIClassRP2040(SPI_X, TFT_MISO, TFT_CS, TFT_SCLK, TFT_MOSI);
|                                                                  ^~~~~~~~
|                                                                  TFT_CS
D:\Code\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_RP2040.c:18:76: error: 'TFT_MOSI' was not declared in this scope; did you mean 'TFT_MISO'?
18 |     SPIClassRP2040 spi = SPIClassRP2040(SPI_X, TFT_MISO, TFT_CS, TFT_SCLK, TFT_MOSI);
|                                                                            ^~~~~~~~
|                                                                            TFT_MISO
```

So the definitions for `TFT_SCLK` and `TFT_MOSI` are missing. I examined the associated header files and c files, and couldn't find their definitions anywhere.

**Workaround I am using:**

To solve the problem, I added the missing definitions in the **User Setup** file as you can see above, and the code compiled just fine.

**PS:** Even with that workaround I am not able to get anything displayed on the display. I was able to use the same configuration and LCD to get it working with an ESP32 board before (with ESP32 SPI). So the LCD configuration is correct. But I will post this issue separately once I am sure everything on my side are correctly setup, such as wiring.

vishnumaiea, 2022-10-24:
I just replaced the release version of TFT_eSPI with the master branch and I can confirm that the issue is unresolved in the latest version.

vishnumaiea, 2022-10-24:
The LCD now works. There was a breakage on the CS line of the LCD.

Bodmer, 2022-10-24:
Yes, you have to define those pins in the setup, they are needed for all processors except the ESP8266 which only has one SPI port.

Here are the pins defined in an [example setup file](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup60_RP2040_ILI9341.h#L124-L130) for the RP2040 with an ILI9341 display.

The library by default uses the SPI 0 port. Since you are using the SPI 1 port so this needs to be defined in the setup file as [noted here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup60_RP2040_ILI9341.h#L124-L130).

Ah, I see you have fixed while I was typing!

vishnumaiea, 2022-10-24:
Maybe you can include a short note on the need for extra pin definition in the user setup file, just like you already have.

closed 2022-10-24 by vishnumaiea