Bodmer/TFT_eSPI #2141: Text horizontally shifted when printing with smooth font at negative x position

tcroon, 2022-11-06:
Hello,
I ran into an issue when printing smooth font text starting at a negative x position. The text is sometimes shifted one or a few pixels to the left or right, which leads to skips if you try to scroll text into negative x positions this way. For scrolling text like this, I finally used a larger sprite which is completely shifted to the left on pushSprite(), but I think that's not the perfect solution with regards to memory consumption.

I tested it with and without sprites and with either setting the cursor position and then printing (println() / printToSprite()) or drawString(). It's always the same result.
I also tried with the standard font (by just using tft.setTextSize(2)) and there everything looked fine. Haven't tested it with unsmooth custom fonts. The smooth font (NotoSansBold15) that I use in the attached example was taken from examples\Smooth Fonts\FLASH_Array\Font_Demo_1_Array.

See below the code for a full working example (for a 160x128 screen but should work with others as well) showing the behaviour (especially visible when x is around -160 to -190 since you then see the second text come scrolling in smoothly). The first scrolling line is printed directly to tft, the second one is printed to a sprite with screen width. These two show the described behaviour. The third scrolling line is using a larger sprite and moves smoothly.

I'm using Arduino IDE with TFT_eSPI V2.4.72 for an ESP32 with a 160x128Px 1,77" SPI TFT display (ST7735).

Here's the code (and I also attached the project as a zip-File):

```
#include <TFT_eSPI.h>
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();
TFT_eSprite sprX = TFT_eSprite(&tft);
TFT_eSprite sprNarrow = TFT_eSprite(&tft);
TFT_eSprite sprWide = TFT_eSprite(&tft);

int x = 0;
String scrollText = "text too long to fit on screen";

// distance between first and second text
int distance = 50;

int textWidth;

#include "NotoSansBold15.h"

void setup(void) {  
  tft.init();
  tft.setRotation(1);
  tft.setTextWrap(false, false);

  // captions directly on tft
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setTextSize(1);
  tft.setCursor(0, 0);
  tft.println("x:");
  tft.setCursor(0, 30);
  tft.println("without sprite:");
  tft.setCursor(0, 60);
  tft.println("narrow sprite (w/ neg. x):");
  tft.setCursor(0, 90);
  tft.println("wide sprite (w/o neg. x):");

  // sprite for showing x value
  sprX.createSprite(50, 20);
  sprX.setTextWrap(false, false);
  sprX.setTextColor(TFT_WHITE, TFT_BLACK);
  
  // narrow sprite with screen width
  sprNarrow.createSprite(160, 20);
  sprNarrow.setTextWrap(false, false);
  sprNarrow.setTextColor(TFT_WHITE, TFT_BLACK);

  // wide sprite to stay in positive x-range when printing (shift whole sprite on pushSprite)
  sprWide.createSprite(600, 20);
  sprWide.setTextWrap(false, false);
  sprWide.setTextColor(TFT_WHITE, TFT_BLACK);

//  // for textSize=2:
//  tft.setTextSize(2);
//  sprX.setTextSize(2);
//  sprNarrow.setTextSize(2);
//  sprWide.setTextSize(2);
  
  // for NotoSansBold15:
  tft.loadFont(NotoSansBold15);
  sprX.loadFont(NotoSansBold15);
  sprNarrow.loadFont(NotoSansBold15);
  sprWide.loadFont(NotoSansBold15);

  textWidth = tft.textWidth(scrollText);
}

void loop() {
  // print x value
  sprX.fillSprite(TFT_BLACK);
  sprX.setCursor(0,0);
  sprX.println(x);
  sprX.pushSprite(0, 10);
  
  // without sprite
  tft.fillRect(0, 40, 160, 20, TFT_BLACK);
  tft.setCursor(x, 40);
  tft.println(scrollText);

  
  // narrow sprite
  sprNarrow.fillSprite(TFT_BLACK);
  
  // Test 1: setCursor and print
  sprNarrow.setCursor(x, 0);
  // Test 1a: println()
  sprNarrow.println(scrollText);
//  // Test 1b: printToSprite()
//  sprNarrow.printToSprite(scrollText);

//  // Test 2: drawString
//  sprNarrow.drawString(scrollText, x, 0);

  
  // wide sprite (x shifted by 440px)
  sprWide.fillSprite(TFT_BLACK);
  sprWide.setCursor(x + 440, 0);
  sprWide.println(scrollText);

  
  // scroll in second text
  if(x < -(textWidth + distance) + 160){
    // without sprite
    tft.setCursor(x + (textWidth + distance), 40);
    tft.println(scrollText);
  
    // narrow sprite
    sprNarrow.setCursor(x + (textWidth + distance), 0);
    sprNarrow.println(scrollText);
  
    // wide sprite (x shifted by 440px)
    sprWide.setCursor(x + 440 + (textWidth + distance), 0);
    sprWide.println(scrollText);
  }
  
  
  // push sprites (wide sprite is now shifted by 440px to negative)
  sprNarrow.pushSprite(0, 70);
  sprWide.pushSprite(-440, 100);

  
  // update x
  x--;
 
  if(x < -textWidth){
    // first text is out, move x to second text
    x += (textWidth + distance);
  }
  

  // 40ms delay for about 25fps
  delay(40);
}
```
[scrollTest.zip](https://github.com/Bodmer/TFT_eSPI/files/9946092/scrollTest.zip)

https://user-images.githubusercontent.com/9355980/200177620-0395d041-d215-4d82-82a3-a3f21b979ebf.mp4

Bodmer, 2022-11-07:
Thanks for reporting this, I have reproduced the problem.

I suspect this may either be linked to the way the library deliberately tries to adust the position for italic fonts (bitmat extends to the left of the character datum) as characters near the x=0 c0ordinate boundary. This is intended to avoid loosing part of a character when plottting at x = 0.

I will investigate further but this looks like an obscure corner case and thus it may take a while for me to find time to look into this in depth. Thus I suggest you stick with the work-around for now.

tcroon, 2022-11-10:
Thank you for the reply, I'm glad it seems it's not just an error in my specific code.

I now did some more small tests with the Sprite.cpp code. What I found out is that the sprite part of my example seems to work as expected if I comment out line 2477 in Sprite.cpp:
`      // if ( cursor_x == 0) cursor_x -= gdX[gNum];
`

Of course I don't know what else would be affected by this (there certainly is a reason for this line). I guess this might be exactly what you meant by the adjustment for italic fonts.

Bodmer, 2022-11-10:
Yes, that is the adjustment I mentioned.

I will need to think how your use case can be handled. It may need a configuration switch otherwise functions like printToSprite will behave diffrently and clip some (typically itallic) characters.

autodiag, 2024-05-07:
is it possible to change the dynamic test that is scrolled while the program is running ? I have tried several variations but unfortunately I am still not successful :/