Bodmer/TFT_eSPI #2143: TFT_eSprite::print() - Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.

PospelovDaniil, 2022-11-06:
Hi Bodmer. 
I think I found a problem with TFT_eSprite, with the print() function, it causes ESP32 to constantly reboot with the error
"Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled."
Please tell me if it's the library error or my error of using it or something else? 
This is a critical error for my project. :(

1. result print("Hello") in an inherited class from the TFT_eSprite - Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled. And with original TFT_eSprite too.
2. Arduino IDE 2.0.1 (latest)
3. TFT_eSPI library version 2.4.72 (latest)
4. ESP32 v2.0.5 (latest)
5. ESP32
6. TFT driver ILI9488 (with touch, in user setup select file: setup21). [https://aliexpress.ru/item/1005003033844928.html?spm=a2g2w.orderdetails.0.0.670c4aa6rUoxj5&sku_id=12000023355126442&_ga=2.26692605.1160083278.1667770563-1500589039.1652675838](url)
7. SPI

Some board configuration parameters:
cpu freq 240mhz
evenst run on core 1
arduino run on core 1
partition scheme Huge app (with default 4mb spiffs dont work too)
flash size 4mb

setup21:
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK  18
#define TFT_CS      15
#define TFT_DC      2
#define TFT_RST     4
#define TOUCH_CS 5

Code for repeat error:
```
#include <SPI.h>
#include <TFT_eSPI.h>

class Widget: public TFT_eSprite
{
public:
  Widget() = delete;
  Widget(TFT_eSPI *tft): TFT_eSprite(tft) {}
  virtual ~Widget() = default;

  //...
};
class wMeteo: public Widget
{
public:
  wMeteo() = delete;
  wMeteo(TFT_eSPI *tft): Widget(tft) {}
  ~wMeteo() = default;

  // ...
};

auto tft = TFT_eSPI();

void setup()
{
  tft.init();
  tft.setRotation(2);
  tft.setCursor(0, 0);
  tft.fillScreen(TFT_BLACK);

  // work
  tft.print("Hello");

  // dont work
  auto originalSpite = TFT_eSprite(&tft);
  originalSpite.createSprite(50, 20);
  originalSpite.fillSprite(TFT_GREEN);
  //originalSpite.print("Hello orignal sprite"); // problem here, uncomment this for see
  originalSpite.pushSprite(0, 30);

  // dont work
  wMeteo meteo(&tft);
  meteo.createSprite(50, 20);
  meteo.fillSprite(TFT_RED);
  //meteo.print("meteo"); // problem here, problem here, uncomment this for see
  meteo.pushSprite(0, 60);
}

void loop()
{

}
```

Bodmer, 2022-11-07:
I have run the sketch and it does not crash. It runs fine and carries on looping in loop().

I guess there is something different in your setup or environment.

Bodmer, 2022-11-07:
Just went to save the sketch and see there are lines that have to be uncommented to cause a crash. When uncommented there is indeed a crash.

I suspect this is a class scope problem. This works fine:
```
#include <SPI.h>
#include <TFT_eSPI.h>

class Widget: public TFT_eSprite
{
  public:
    Widget() = delete;
    Widget(TFT_eSPI *tft): TFT_eSprite(tft) {}
    virtual ~Widget() = default;

    //...
};
class wMeteo: public Widget
{
  public:
    wMeteo() = delete;
    wMeteo(TFT_eSPI *tft): Widget(tft) {}
    ~wMeteo() = default;

    // ...
};

auto tft = TFT_eSPI();

TFT_eSprite originalSpite = TFT_eSprite(&tft);
wMeteo meteo(&tft);

void setup()
{
  tft.init();
  tft.setRotation(2);
  tft.setCursor(0, 0);
  tft.fillScreen(TFT_BLACK);

  // work
  tft.print("Hello");

  // dont work
  originalSpite.createSprite(50, 20);
  originalSpite.fillSprite(TFT_GREEN);
  originalSpite.print("Hello orignal sprite"); // problem here, uncomment this for see
  originalSpite.pushSprite(0, 30);

  // dont work
  meteo.createSprite(50, 20);
  meteo.fillSprite(TFT_RED);
  meteo.print("meteo"); // problem here, problem here, uncomment this for see
  meteo.pushSprite(0, 60);
}

void loop()
{
  tft.print("Hello");
  delay(1000);
}
```

Bodmer, 2022-11-07:
Making the classes static will also work:
```
void setup()
{
  tft.init();
  tft.setRotation(2);
  tft.setCursor(0, 0);
  tft.fillScreen(TFT_BLACK);

  // work
  tft.print("Hello");

  // dont work
  static auto originalSpite = TFT_eSprite(&tft);
  originalSpite.createSprite(50, 20);
  originalSpite.fillSprite(TFT_GREEN);
  originalSpite.print("Hello orignal sprite"); // problem here, uncomment this for see
  originalSpite.pushSprite(0, 30);

  // dont work
  static wMeteo meteo(&tft);
  meteo.createSprite(50, 20);
  meteo.fillSprite(TFT_RED);
  meteo.print("meteo"); // problem here, problem here, uncomment this for see
  meteo.pushSprite(0, 60);
}
```

PospelovDaniil, 2022-11-07:
Thanks for the help, yes, it works now.

closed 2022-11-07 by PospelovDaniil