Bodmer/TFT_eSPI #2144: Using TFT_eSPI with an ESP8266

karstengit, 2022-11-07:
I already used this wonderful library successful with an STM32 Blue-Pill.
Now i am trying to connect this [display](http://www.lcdwiki.com/2.8inch_SPI_Module_ILI9341_SKU:MSP2807) to an ESP8266-12F on an own small adapter PCB.

I am using again the benchmark test that can be find in the examples.

```
/*
 Adapted from the Adafruit and Xark's PDQ graphicstest sketch.
	Connections of the display
	1 VCC		3V3
	2 GND		GND			
	3 CS		GPIO15		
	4 Reset		RST	
	5 D/C		GPIO16
	6 MOSI (SDI)	GPIO13	
	7 SCK		GPIO14
	8 LED		GPIO0
	9 MISO (SDO)	GPIO12
*/

#include "SPI.h"
#include "TFT_eSPI.h"

// Use hardware SPI
TFT_eSPI tft = TFT_eSPI();

void setup() {
	// Setting the mode of the pins
	pinMode(0, OUTPUT);			// TFT LED Light
	digitalWrite(0, 1);			// LED on
	
	tft.init();
}

void loop(void)
{
	Serial.println(F("Benchmark                Time (microseconds)"));

	uint32_t usecHaD = testHaD();
	Serial.print(F("HaD pushColor            "));
	Serial.println(usecHaD);
	delay(100);
...
```

with this User_Setup.h in the same folder

```
// Only define one driver, the other ones must be commented out
#define ILI9341_DRIVER

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D0  // Data Command control pin
// #define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

// #define TFT_BL PIN_D0  // LED back-light (only for ST7789 with backlight control pin)

#define TOUCH_CS PIN_D2     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000
```

It compiles, can be flashed and the UART is working, but there cannot be seen any signal on the connected lines with an oscilloscope.
What i am doing wrong?

karstengit, 2022-11-07:
Sorry for opening as bug report - I did not see that there is an active discussion forum.
https://github.com/Bodmer/TFT_eSPI/discussions/2146

Bodmer, 2022-11-07:
I suspect this is a pin mapping problem. The library was originally developed soley for the ESP8266 NodeMCU board and thus the default pin definitoons are for that pasticular boaard.

Unfortunately different board suppliers map different "Dxx" pins to different GPIO, so in your instance it would be best to use direct GPIO numbers.

Just for information only,  this is the mapping from PIN_Dxx definitions to GPIO for the default:

```
// These are the pins for ESP8266 boards
//      Name   GPIO    NodeMCU      Function
#define PIN_D0  16  // GPIO16       WAKE
#define PIN_D1   5  // GPIO5        User purpose
#define PIN_D2   4  // GPIO4        User purpose
#define PIN_D3   0  // GPIO0        Low on boot means enter FLASH mode
#define PIN_D4   2  // GPIO2        TXD1 (must be high on boot to go to UART0 FLASH mode)
#define PIN_D5  14  // GPIO14       HSCLK
#define PIN_D6  12  // GPIO12       HMISO
#define PIN_D7  13  // GPIO13       HMOSI  RXD2
#define PIN_D8  15  // GPIO15       HCS    TXD0 (must be low on boot to enter UART0 FLASH mode)
#define PIN_D9   3  //              RXD0
#define PIN_D10  1  //              TXD0

#define PIN_MOSI 8  // SD1          FLASH and overlap mode
#define PIN_MISO 7  // SD0
#define PIN_SCLK 6  // CLK
#define PIN_HWCS 0  // D3

#define PIN_D11  9  // SD2
#define PIN_D12 10  // SD4
```

I think these settings may be suitable for your board if it is wired like this. Note that the SPI ports are mapped to specific GPIO pins on the ESP8266 and thus MISO, MOSI and SCLK must use those pins.

```
#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK  14
#define TFT_CS     15  // Chip select control pin
#define TFT_DC      0  // Data Command control pin
#define TFT_RST     2  // Reset pin (could connect to RST pin)
```

Note also that some pins must be at a certain level at boot time for the processor to startup and run the sketch (see comments above). Typically ready built board bias these pins into the correct state.

Run the Test and diagnostic example "Read_User_Setup" and look at the serial monitor output to check the compiler is picking up the settings correctly. Post the output here with a copy of your setup file if you are not sure the output is correct.