Bodmer/TFT_eSPI #2148: image pushToSprite combined with background setSwapBytes(true) changes image Sprite colour

NDC-SC, 2022-11-07:
When a Sprite created as a shape is pushed onto a background Sprite, the colour of the shape Sprite is blue when defined as TFT_RED, is green when defined as blue or is red when defined as green, when the back.setSwapBytes(true) instruction is included.

The colour of the shape Sprite is correct with the back.setSwapBytes(false) instruction, but then the colours of the background Sprite are distorted.

A TTGO T-Display V1.1 was used with  #include <User_Setups/Setup25_TTGO_T_Display.h> in the User_Select_Setup file

#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI(); 
TFT_eSprite back = TFT_eSprite(&tft);  // background Sprite
TFT_eSprite img = TFT_eSprite(&tft);   // shape Sprite
#include "background.h"  // contains : const unsigned short backgnd[32400] PROGMEM={....
void setup()
{
  tft.init();
  tft.setRotation(1);
  back.createSprite(240, 135);    // background Sprite
  back.setSwapBytes(true);      // require setSwapBytes for image colours
  img.createSprite(28, 28);
  back.pushImage(0, 0, 240, 135, backgnd);  // load backgnd file
  img.fillCircle(14, 14, 14, TFT_RED);
  img.fillCircle(14, 14, 9, TFT_BLACK);  // create transparent effect for centre area of shape Sprite
  img.pushToSprite(&back, 120, 60, TFT_BLACK);  // push shape Sprite to background Sprite
  back.pushSprite(0,0);
}
void loop()  {}

Bodmer, 2022-11-07:
Post a complete sketch. I suspect the image is byte swapped. This can be handled but I need to see a complete sketch with a representative bitmap image.

Bodmer, 2022-11-08:
I had a look at the sketch and ran it.

The background image is byte swaped so swap bytes for the destination sprite is set to true so the colours are correct in the rendered sprite.

The problem is that a sprite is also an image but when pasting a sprite into another one the destination sprite code will swap the bytes if swap bytes is still true (which is not wanted in this case).

The solution is to set the "back" sprite byte swap setting back to the default case of "false" after pushing the backgnd image thus:

```
  back.setSwapBytes(true);      // Tell sprite class to swap bytes for this image
  back.pushImage(0, 0, 240, 135, backgnd);
  back.setSwapBytes(false);    // Change back to default "false" so image bytes are not swapped
```

I have tested this and it works (red circle drawn).

closed 2022-11-08 by Bodmer

NDC-SC, 2022-11-08:
Thanks very much.
It appears that the RGB colours were being shifted one position to GBR, so I was setting the image colour accordingly to end up with the required colour.
Your suggestion is much better.