Bodmer/TFT_eSPI #2159: ESP32 TFT ST7789 display not showing

audreylim, 2022-11-10:
I'm trying to get the animated eyes example, or any example here, show up on my TFT display. However, nothing shows up.

Background:
- Processor: ESP32-DEVKITC-v4
- Display: [ 1.54 inch LCD module 240 x 240 pixels. ](https://www.waveshare.com/1.54inch-lcd-module.htm)
- Driver: ST7789 
- Interface: SPI
- IDE: Arduino

The following is the ESP32 - TFT connection:

VCC - 3v3
GND - GND
DIN - 23
CLK - 18
CS - 15
DC - 2
RST - 4
BL - not connected

[image]
[image]

Setup file:
[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/9977906/User_Setup.zip)

```
[code]
TFT_eSPI ver = 2.4.42
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 240
Display height = 240

MOSI    = GPIO 23
SCK     = GPIO 18
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
[/code]
```

TFT_ReadWrite_Test example's output in serial monitor:

```
 ERROR                 ^^^^
 Pixel value written = 4000
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 8000
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 1
 Pixel value read    = 0
 ERROR                 ^^^^
```

Result when running animated eyes example:

[image]

Bodmer, 2022-11-10:
I looked at your posted setup file and TFT_CS is set to -1, it should be 15.

Bodmer, 2022-11-14:
I will assume this is resolved. If not please post back.

closed 2022-11-14 by Bodmer

audreylim, 2022-11-21:
That was it. Fixed, thank you!